Pi4J/pi4j-v1 #248: mvn install pi4j compile  error

hankp46, 2016-06-29:
I'm trying to enable NonPrivileged Access in a Java app so I can run without sudo. I added the following code:
 private void addActionField() {
        final RecordingRMSListener rmsListener = this;
        actionButton = new JButton(START_LABEL);
        actionButton.setEnabled(true);

        GpioUtil.enableNonPrivilegedAccess();

	final GpioController gpio = GpioFactory.getInstance();

        final GpioPinDigitalInput myButton = gpio.provisionDigitalInputPin(RaspiPin.GPIO_02, PinPullResistance.PULL_DOWN);
        myButton.addListener(new GpioPinListenerDigital() {
            @Override
            public void handleGpioPinDigitalStateChangeEvent(GpioPinDigitalStateChangeEvent event){
                actionButton.doClick();
            }
        });
I get the following error:
 cannot find symbol
[ERROR] symbol:   method enableNonPrivilegedAccess()
[ERROR] location: class com.pi4j.wiringpi.GpioUtil

hankp46, 2016-06-29:
Operator error. Did not read far enough. Forgot to update Pom File. Everything working perfectly!

closed 2016-06-29 by hankp46

savageautomate, 2016-06-29:
Glad you got it working :-)