Pi4J/pi4j-v1 #252: ServoBlaster/servod support for physical pins 27-40

salk31, 2016-07-15:
I tried to use physical pin 29. ServoBlaster seems happy with this but RPIServoBlasterProvider blows up because it doesn't have mappings such as:
        RPIServoBlasterProvider.definePin(RaspiPin.GPIO_21, "P1-29");
        RPIServoBlasterProvider.definePin(RaspiPin.GPIO_22, "P1-31");

I've tested this (hacking it in with package scope) and it seems to work for me.

If this is really a bug would you be up for me submitting a patch?

natdan, 2016-07-15:
Is it possible that it doesn't work on some versions of RPi?
Definitively worth submitting a patch!
(In the meantime I'll remind myself what happened there and why those are not provided)

salk31, 2016-09-20:
I think I can see how to turn on/off the extra pins based on type/model of board.

Could/should RPIServoBlasterProvider support other numbering schemes? It already translates between WiringPI and its own naming schema so not a new layer of abstraction. If so would you accept a patch with both together?

closed 2017-03-06 by salk31