Pi4J/pi4j-v1 #253: Add support for BananaPi M3

616slayer616, 2016-07-15:
Hi,
I would like to have Banana Pi M3 support.
So I forked the repository and used the bananapi / bananapro files as template.
I also included wiringPi for the M3 from BPI-SINOVOIP/BPI-WiringPi.

Controlling the pins seems to work (I controlled a fan for testing)
What I did not test yet is using the pins as output.  Moreover I lack the hardware and the experience to test serial, I2C , SPI.

Are you interested in supporting the M3?

Thanks,
Patrick

savageautomate, 2016-07-25:
Yes, I am interested.  I would like to get 1.0.1 fully released and then re-work some of the build logic for these alternate platforms.  Perhaps even consider moving the platform implementation code out to separate impl JARs so the library could be more portable and new platforms implemented without changes to the core JARs.  

Let's revisit this soon after I get 1.0.1 fully released. 

Thanks, Robert

kwaiette, 2017-06-29:
Is there any news on this?  I'm trying to get Pi4j to work on a Banana Pi M3 and so far am unable.  I've looked at 616slayer616's fork to see what I could do to get it working but as far as I can tell, it has been merged into the main project.  I have the BPI-SINOVOIP/BPI-WiringPi working on the M3 but so far haven't been able to figure out how to use it in Pi4j.

616slayer616, 2017-06-30:
There is no news on this. I got it tonwork fairly ok altgough the compiler toolchain made some problems but I can't remember what exactly.
My changes are removed from my repo because I was not happy with it and there where too many changes to the original fork.
But you should be able to get it working based on the other banana pi implementations. Just clone the one for the pro model and adjust it. That is basically what I did. Then you'll probably have to change the toolchain according to the bpi3 wiringpi procedure.

savageautomate, 2017-08-06:
**UPDATE**
I have just added experimental support for the Sinovoip BPI series "BananaPi" variants.

COMMIT: https://github.com/Pi4J/pi4j/commit/92f06587e92d740d78a98ff45881c6ae44604316

This is included in the the latest 1.2-SNAPSHOT build now.

Please note that I do not have any of these devices so I'm not able to really test it.
I have included a new "bpi" platform that you have to choose when starting your program.

```
        // ####################################################################
        //
        // since we are not using the default Raspberry Pi platform, we should
        // explicitly assign the platform as the Sinovoip BPI platform.
        //
        // ####################################################################
        PlatformManager.setPlatform(Platform.BPI);
```

**STATUS:**

- I have added support for building the BPI-WiringPi library (https://github.com/BPI-SINOVOIP/WiringPi) and embedding them inside the Pi4J JARs.  
- I have created a new BpiGpioProvider (https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/BpiGpioProvider.java) and BpiPin (https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/BpiPin.java) class that should be used.
- The BpiPin class is purely a copy from the BananaPro class, I have not verified the pin mappings with the BPI devices. 
- I have not tested anything (other than the build)
- I have not added any samples to the examples yet.
- Board detection (M1, M2, M3, M1+, M2+, M2P, etc) has not been added yet. 

Feel free to start testing it and report back any issue .. or better yet submit pull requests with fixes.

Thanks, Robert

savageautomate, 2021-01-14:
**Closed**; Pi4J v1.4 drops support for alternate platforms to better focus on core functionality.  Pi4J v2.0 will add support for abstracting platforms in a new pluggable architecture.

closed 2021-01-14 by savageautomate