Pi4J/pi4j-v1 #254: Using Pi4J with an ESC & a brushless motor

DarkmatterVale, 2016-07-15:
Is it possible to use Pi4J to initialize an ESC & then run a brushless motor?

More specifically, can I use Pi4J to run a WP-1040 Brushed (Crawler) ESC with a brushless DC motor (http://www.hobbyking.com/hobbyking/store/__26267__133RPM_Brushed_Motor_w_75_1_Gearbox.html)? If so, how would I go about accomplishing that? Any help would be appreciated.

Thank you for the help in advance!

pasha-shestakov, 2016-07-25:
Im just curious, but looks like your building an RC car? (well is guess it probably not remote controlled :)) If not a secret, may i ask what for?

natdan, 2016-07-25:
Yes.

Please refer to: https://www.raspberrypi.org/forums/viewtopic.php?f=81&t=154475&sid=cbda77bf4971b15cf735c0ab51682d19

If driven directly from RPi you can use servoblaster driver (https://github.com/richardghirst/PiBits/tree/master/ServoBlaster). Example is here: https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/RPIServoBlasterExample.java

It should be straightforward.

natdan, 2016-07-25:
@pasha-shestakov  while you are waiting for @DarkmatterVale 's answer - I can give you another: I'll be building a robot with a school's club soon (immediately after summer) and it will hopefully be using similar setup: brushed/brushless car ESC with servoblaster, pi4j, etc... And it *will* be remotely controlled (serial over bluetooth or socket over WiFi from Android phone/web interface).

savageautomate, 2016-12-02:
Questions/conversation should move to the Pi4J forums you may get a larger audience to help.  https://groups.google.com/forum/#!forum/pi4j

Closing this issue.  Thanks, Robert

closed 2016-12-02 by savageautomate