Pi4J/pi4j-v1 #255: Add support for Odroid C2

savageautomate, 2016-07-25:
## Issue

Officially add support for Odroid C2 boards.

## Progress

**Odroid C2**
- [x] source Odroid C2 hardware
- [X] add support for reading analog input pins
- [X] add support for analog input pin events
- [ ] include wiringPi x64 for Odroid in build
- [ ] package wiringPi x64 for Odroid compiled SO libs in JAR
- [ ] test/validate GPIO outputs on Odroid C2
- [ ] test/validate GPIO inputs on Odroid C2
- [ ] test/validate GPIO input listener events on Odroid C2  ((**SEE KNOWN ISSUES**))
- [ ] test/validate analog inputs on Odroid C2
- [ ] test/validate analog input listener events on Odroid C2
- [ ] ~~test/validate  PWM communication on Odroid C2~~  ((**NOT SUPPORTED, SEE KNOWN ISSUES**))
- [ ] test/validate serial communication on Odroid C2
- [ ] test/validate  I2C communication on Odroid C2
- [ ] test/validate  SPI communication on Odroid C2  ((**SEE KNOWN ISSUES**))
- [ ] add project examples for Odroid C2
- [ ] update project readme for Odroid C2
- [ ] update project documentation for Odroid C2

## Summary

**Odroid C2**
Work in progress.  Need to setup cross-compiler toolchain for x64 builds of WiringPi and Pi4J.

## Known Issues

**GPIO Listener Events (edge triggered interrupts)**  **[_APPLIES ONLY TO C1/C1+/C2_]**
Pi4J uses edge detection for GPIO pin state change events.  The Odroid C1/C1+/C2 only permits up to four GPIO pins to be configured with edge detection for both "rising" and "falling" edges (a.k.a., "both").  Thus, you can only use a maximum of four GPIO input pins with listener events.  Alternatively, you can manually poll for GPIO pin state changes.  The included sample code demonstrates both the event listener and polling for pin state changes:  https://github.com/Pi4J/pi4j/blob/develop/pi4j-example/src/main/java/odroid/GpioListenAllExample.java

**PWM**
PWM is not supported in the Hardkernel Odroid WiringPi port.  Thus PWM is not currently supported by Pi4J for the Odroid boards (C1/C1+/C2).   There is no PWM function on the 30pin GPIO header on the XU3/XU4.

```
  if (	piModel == PI_MODEL_ODROIDC  ||
	piModel == PI_MODEL_ODROIDC2 ||
	piModel == PI_MODEL_ODROIDXU_34)  {
	fprintf (stderr, "%s : Unsupport function on %s model\n",
		__func__, piModelNames [piModel]);
	exit (EXIT_FAILURE) ;
  }
```
_See: https://github.com/hardkernel/wiringPi/blob/master/wiringPi/wiringPi.c#L2254_

**WiringPi SoftPWM**
SoftPWM is not supported on all Odroid GPIO pins on the C1/C1+/C2/XU4.  This limitation is due to a GPIO pin mapping issue in the WiringPi port for Odroid by Hardkernel.

**SPI**
SPI modes (other than the default SPI MODE 0) are not supported in the Hardkernel Odroid WiringPi port.  We are expecting the `wiringPiSPISetupMode()` to be implemented in this source file:  https://github.com/hardkernel/wiringPi/blob/master/wiringPi/wiringPiSPI.c to support this feature.

trbLeeciN, 2016-09-07:
Hello,
I am trying to use pi4j for odroid c2. The link "[https://github.com/Pi4J/pi4j/blob/develop/pi4j-example/src/main/java/odroid/GpioListenAllExample.java]" you gave is broken. Can you please give alternative. Or is it same with odroid c1 listener? 
And I want to ask, if I use only rising edges, can I use 8 gpio pins instead of 4. Thank you..

riyal123, 2019-07-22:
Any update for Odroid c2?

savageautomate, 2021-01-12:
**CLOSED**; all new development efforts are in Pi4J v2.x (https://v2.pi4j.com)

closed 2021-01-12 by savageautomate