Pi4J/pi4j-v1 #256: Is there any support for Pimoroni Scroll-pHat ?

cab13140, 2016-07-28:
Hello,
I'm new to Pi, and I don't really know how I2C works with Pi4J
I saw in WiringPI that support was there, but I scrolled all Pi4J repository, and found nothing, exept I2CDisplay...
Is there any direct class or interface for the Scroll-pHat

Thanks

savageautomate, 2016-07-28:
Hi @cab13140 

Here is a basic example using I2C communications via Pi4J on the Raspberry Pi:
https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/I2CExample.java

First you get a handle to a supported I2C bus using the I2CFactory.getInstance() method.
```
// get the I2C bus to communicate on
I2CBus i2c = I2CFactory.getInstance(I2CBus.BUS_1);
```

_(**Note**: you may have to enable the I2C support on the Raspberry Pi via the raspi-config utility under the Advanced options menu before attempting to using it via Pi4J.)_

Then you can create a I2C device instance from your bus object using the device's address on the bus. 
```
// create an I2C device for an individual device on the bus that you want to communicate with
// in this example we will use the default address for the TSL2561 chip which is 0x39.
I2CDevice device = i2c.getDevice(TSL2561_ADDR);
```

Now you can read and write to the I2C device.

You can view the I2C classes here for more details on the available API methods:
https://github.com/Pi4J/pi4j/tree/master/pi4j-core/src/main/java/com/pi4j/io/i2c

Thanks, Robert

closed 2016-07-28 by savageautomate

cab13140, 2016-08-01:
Thanks ! It's Working... but a direct implementation would be easier.

Issue can be closed ! It's working fine, thanks.