Pi4J/pi4j-v1 #257: New Serial JNI interface does not allow baudrates above 230400 ?

florian-z, 2016-07-28:
With the old serial interface (by using .open() directly) it is possible to use baudrates above 230400, e.g.:

    int baudrate = 1000000;
    final Serial serial = SerialFactory.createInstance();
    serial.open(Serial.DEFAULT_COM_PORT, baudRate);

But I cannot find a way to do this with the new serial interface (using .baud() to set the baudrate).

Is this on purpose?

savageautomate, 2016-07-28:
You  an still use the Serial.open() method to pass a custom baud rate as an integer.
https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/serial/Serial.java#L102

The SerialConfig object is just an optional convenience using the included enums. 

Thanks, Robert

closed 2016-07-28 by savageautomate

florian-z, 2016-07-28:
Okay, great.