Pi4J/pi4j-v1 #258: Frequent Array Out Of Bounds Exception

NewLunarFire, 2016-08-03:
Hi,

I'm using Pi4j 1.1-SNAPSHOT on a Raspberry Pi 2 B+ model with Raspbian and OpenJDK 8. For a few days (starting August 1st), I started to receive a number of out of bounds exception. Those exceptions are not originating from my own code, as you can see in the stack trace below, but from the Pi4J serial port listener.

I don't know if a frequent change occurred in 1.1-SNAPSHOT that causes those exceptions. It doesn't affect us too much but could be problematic in other cases.

```
Exception in thread "Thread-5" java.lang.ArrayIndexOutOfBoundsException
    at java.lang.System.arraycopy(Native Method)
    at com.pi4j.io.serial.impl.SerialByteBuffer.resize(SerialByteBuffer.java:107)
    at com.pi4j.io.serial.impl.SerialByteBuffer.write(SerialByteBuffer.java:123)
    at com.pi4j.io.serial.impl.SerialByteBuffer.write(SerialByteBuffer.java:116)
    at com.pi4j.io.serial.impl.SerialImpl$1.onDataReceive(SerialImpl.java:179)
    at com.pi4j.jni.SerialInterrupt.onDataReceiveCallback(SerialInterrupt.java:110)
```

savageautomate, 2016-08-07:
@NewLunarFire ,

Can you retry this with the v1.1 release build?  
If it still occurs, can you provide a snippet of example code that causes this error?

I'll look back in the history to see if something changes recently with the serial impl.
_(I typically use the Oracle JVM for development and testing, but hopefully that is not an issue.)_

Thanks, Robert

NewLunarFire, 2016-08-08:
I'm using a SerialDataEventListener and in my callback function, I fetch the data using SerialDataEvent.getBytes()

Code that creates the serial port:
```Java
// Create a Serial Port Instance
port = SerialFactory.createInstance();
		
//Open Specified port
port.open(portName, BAUDRATE);
		
//Add this as an event listener
port.addListener(this);
```

Code that reads data from the serial port:
```Java
try {
	listener.receive(event.getBytes());
} catch (IOException e) {
	logger.warn("Error reading serial port data", e);
}
```

I was using OpenJDK 8 but we decided to switch to Oracle JDK 8.

I was using the snapshot version because 1.1 had no release yet and I wanted unprivileged GPIO access. I'm going to switch to the release and warn if I ever see the issue again.

closed 2016-12-02 by savageautomate

Abrasam, 2017-02-05:
I still have this issue occurring randomly, sometimes it will even occur on and off during the running of a program..

felislynx-silae, 2020-08-17:
@savageautomate This issue still persist in latest version.