Pi4J/pi4j-v1 #262: I2C Bus doesnt need locks

colinrgodsey, 2016-08-03:
* I2CBus instances are cached globally, forever? One per bus.
* I2CBusFactory gladly serves closed buses. 
* I2CBus#close method present, but if its meant to stay open forever, why close it? Exposes side-effects to other users of the bus.
* I2C file descriptors provide kernel-level locking/synchronization. Parallel access to a single FD isn't good, but parallel access to multiple FDs for the same bus is just fine. 
* Allows user to open/close bus/FD just like normal FD based resources (files, sockets, etc)
* Tighter I2C timing as the kernel has its own queue of operations to work from.

rlsutton1, 2016-08-03:
Are you suggesting by-passing wiring pi? ... I've often thought that would be a good way to go, but haven't convinced myself its possible.

colinrgodsey, 2016-08-03:
hmm Im not sure about the wiring-pi i2c impl, just the native pi4j impl (which uses i2c-dev, ioctl, and r/w file descriptors)

natdan, 2016-08-04:
Does your pull request include those changes or you're proposing we remove existing synchronisation from I2C bus impl?

colinrgodsey, 2016-08-04:
@natdan my current PR doesnt address this- those changes were isolated enough I could make the update and keep the same sync model. Can address that in another PR, wanted to get more input though.

I do however have a hack currently that lets me get multiple I2CBus instances for the same bus (and thus dont share locking) and its worked fine for a few days on my sensor box.

colinrgodsey, 2016-08-04:
alright, so my plan for this is.... leave the locks in the I2CBusImpl. If somebody is intentionally sharing these across threads, I want to keep that compatibility and not have this update break everything down the line for some people. 

I'm just going to remove the locked 'bus cache'  and return a new bus impl for each request. This way a bus can still be shared across threads fine, but any implementations requesting a bus from the provider will each get a new instance with its own file descriptor. They can still run like normal with the same consistency guarantees, but they will not have any contention over user-space locks.

colinrgodsey, 2016-08-05:
I'm just going to wrap this fix in with my last branch and make a new PR. They make lots of changes to the same files, so wanted to avoid a merge later.

natdan, 2016-10-09:
Sorry for leaving it aside for this long. I'm setting up everything on my desk again and will be trying it out very soon. In the mean time I've got an idea how to slowly incorporate in i2c and keep everyone happy! Will write more as soon as I deploy on my RPi3!

natdan, 2016-11-11:
Before we close this (and https://github.com/Pi4J/pi4j/issues/260) could you, please, make a pull request with your LinuxFile? Current pull request has stopped at impl using "SharedSecrets".

colinrgodsey, 2016-11-11:
Yes definitely! I got held up with the Linux file implement I'm using because I couldn't think of a graceful way to handle unmapping mmap bytebuffers. I'll get up what I have tho, has been in 24/7 production in my device since I last talked about it (several i2c devices, same bus)

colinrgodsey, 2016-11-14:
Created PR with the new impl: #289

closed 2017-02-16 by colinrgodsey