Pi4J/pi4j-v1 #266: Exception in thread "main" java.lang.NoClassDefFoundError: com/pi4j/io/gpio/GpioFactory

zfeng44, 2016-08-09:
I am using 
      <dependency>
          <groupId>com.pi4j</groupId>
          <artifactId>pi4j-core</artifactId>
          <version>1.1</version>
      </dependency>

pi@raspberrypi:~/Desktop/LED-Weather-Inidcator $ java -jar target/LED-Weather-Inidcator-0.0.1-SNAPSHOT.jar
Exception in thread "main" java.lang.NoClassDefFoundError: com/pi4j/io/gpio/GpioFactory
	at main.java.SensorObjects.PirMotionSensor.<init>(PirMotionSensor.java:70)
	at main.java.SensorObjects.PirMotionSensor.main(PirMotionSensor.java:63)
Caused by: java.lang.ClassNotFoundException: com.pi4j.io.gpio.GpioFactory
	at java.net.URLClassLoader$1.run(URLClassLoader.java:366)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	... 2 more

616slayer616, 2016-08-16:
How did you build your jar?
Sounds like you did not build the dependencies into the artifact.

savageautomate, 2021-01-14:
Closed; no further activity

closed 2021-01-14 by savageautomate