Pi4J/pi4j-v1 #268: ConcurrentModificationException on pinStateChangeCallback

wnojunior, 2016-08-17:
When i try call removeAllListners() on callbackup function this exception is lunched. Maybe this function is try to remove a item of a List during an element iteration.

luejerry, 2016-09-01:
I occasionally get this issue as well since updating from 1.0 to 1.1. I reverted back to 1.0 using the same code and do not encounter this. The stacktrace I get is

```
Exception in thread "Thread-55" java.util.ConcurrentModificationException
        at java.util.ArrayList$Itr.checkForComodification(ArrayList.java:901)
        at java.util.ArrayList$Itr.next(ArrayList.java:851)
        at com.pi4j.io.gpio.GpioProviderBase.dispatchPinDigitalStateChangeEvent(GpioProviderBase.java:354
)
        at com.pi4j.io.gpio.WiringPiGpioProviderBase.pinStateChange(WiringPiGpioProviderBase.java:290)
        at com.pi4j.wiringpi.GpioInterrupt.pinStateChangeCallback(GpioInterrupt.java:128)
```

savageautomate, 2016-12-02:
I was not able to reproduce the problem using version 1.1.  
Here is my sample/test code:

```
        // create gpio controller
        final GpioController gpio = GpioFactory.getInstance();

        // provision gpio pin #02 as an input pin with its internal pull down resistor enabled
        final GpioPinDigitalInput myButton = gpio.provisionDigitalInputPin(RaspiPin.GPIO_02, PinPullResistance.PULL_DOWN);

        // set shutdown state for this input pin
        myButton.setShutdownOptions(true);

        // create and register gpio pin listener
        myButton.addListener(new GpioPinListenerDigital() {
            @Override
            public void handleGpioPinDigitalStateChangeEvent(GpioPinDigitalStateChangeEvent event) {
                // display pin state on console
                System.out.println(" --> GPIO PIN STATE CHANGE: " + event.getPin() + " = " + event.getState());

                myButton.removeAllListeners();
            }
        });
```

It looks like the listeners collection was cloned before attempting to iterate over them and call each listener.  See https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/wiringpi/GpioInterrupt.java#L119

But maybe something else is going on, let me know if you have some reproducible test code to try out to narrow down where this exception is coming from.   Thanks!

closed 2016-12-02 by savageautomate

gramm0001, 2017-03-30:
FYI, I can't reproduce it on demand, but one out of every seven times I start my program, which is using the 1.2 SNAPSHOT, I get:

### Exception in thread "Thread-4" java.util.ConcurrentModificationException
	at java.util.ArrayList$Itr.checkForComodification(ArrayList.java:901)
	at java.util.ArrayList$Itr.next(ArrayList.java:851)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:550)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:538)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:639)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:667)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:661)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:684)
	at MainClasses.CheckDoorStatus.run(CheckDoorStatus.java:33)

albahrani, 2017-03-30:
It looks like the access to `listeners ` in `GpioProviderBase` is only partially synchronizing, the dispatch - methods are not.
In `GpioControllerImpl`  access to `pins `ArrayList is completely unsynchronized, if concurrency is not handled somewhere else the problem is the missing synchronizing there.

I was thinking about a solution not removing from the `pins `ArrayList at all, but changing the state within when provisioning. This idea is based on the assumption that the maximum amount of Pins in memory is limited by hardware and each Software will use a somehow static subset of it. Then synchronizing the change of each Pin's state would be sufficient and synchronizing the `pins `ArrayList would not be needed at all. Okay, only modification of pins would be when initially adding the Pin. So, synchronizing this would be all.