Pi4J/pi4j-v1 #269: ArrayIndexOutOfBoundsException at com.pi4j.io.serial.impl.SerialByteBuffer.resize(SerialByteBuffer.java:107)

jasonjKang, 2016-08-19:
I am using 1.1 released for serial communication.

I got below exception;
Exception in thread "Thread-70707" java.lang.ArrayIndexOutOfBoundsException
       	at java.lang.System.arraycopy(Native Method)
       	at com.pi4j.io.serial.impl.SerialByteBuffer.resize(SerialByteBuffer.java:107)
       	at com.pi4j.io.serial.impl.SerialByteBuffer.write(SerialByteBuffer.java:123)
       	at com.pi4j.io.serial.impl.SerialByteBuffer.write(SerialByteBuffer.java:116)
       	at com.pi4j.io.serial.impl.SerialImpl$1.onDataReceive(SerialImpl.java:179)
       	at com.pi4j.jni.SerialInterrupt.onDataReceiveCallback(SerialInterrupt.java:110)

SerialByteBuffer.java:107 is; 
System.arraycopy(buffer, readIndex, new_buffer, 0, writeIndex);

but, what if (readindex + writeindex > buffer.length) ?
impossible?

jonas-lauber, 2017-01-22:
Yep, same issue here... :-(

pundliksarafdar, 2017-11-22:
I am running Serial Reception on every 10 seconds after 10 to 12 times I am getting this exception

Bradeskojest, 2018-06-02:
We have problem with this exact thing as well. 
At the moment I believe it can be easily replicated by 2 threads trying to read/write to the port. The problem is, that this happens even if this is `synchronized{}`

Example:
```
 public synchronized byte[] writeRead(byte[] bytes, int timeout){
 long startTime = System.nanoTime();
        long time;

        this.serialPort.write(bytes);
        do {
            try {
                Thread.sleep(10);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
            long now = System.nanoTime();
            time = (now - startTime)/1000000;
        } while (this.serialPort.available() == 0 || time >= timeout);

        if(this.serialPort.available() != 0){
            byte[] response = this.serialPort.read();
            if (logger.isTraceEnabled()) {
                logger.trace("Thread: {}, COM port {}: writeread. Read: '{}'",
                        Thread.currentThread().getName(),this.serialConfig.device(),
                        new String(response));
            }
            return response;
        }
...
}
```

closed 2021-02-17 by savageautomate

savageautomate, 2021-02-17:
I realize this is an old issue but I was reviewing prior to v1.4 release and believe I had discovered the problem.  The fix is included here: https://github.com/Pi4J/pi4j/commit/2570dceeb13dc0375da57bf2a1ee3f8299fec908

The serial data receive buffer is intended to be a circular buffer; however the resizing function did not properly take into consideration the circular nature and thus there were cases upon resizing the buffer that data could get lost and index pointers would be set incorrectly.

commit 78b1d16b (2021-02-17): Fix #269; ArrayIndexOutOfBoundsException at SerialByteBuffer.resize()
commit 2570dcee (2021-02-17): Fix #269; ArrayIndexOutOfBoundsException at SerialByteBuffer.resize()