Pi4J/pi4j-v1 #271: Concurrency issue with blink() in MCP23S17

tomtibbetts, 2016-08-25:
I have an issue with the blink function while using the MCP23S17.  What is happening is that the end state of the digital output is not always what I would expect it to be.  This is an intermittent but frequent problem.  I think it happens for pulse as well but more rarely.  I'm including sample code for your review.

Here's the scenario.  The code emulates a control panel of a small model railroad yard.  There are five active low push buttons, four bi-directional stall motors, and four bi-directional LEDs.  When you push one of the buttons, it will execute a small portion of code to properly align the stall motors in the yard.  While the motors are moving, the bi-directional LEDs blink for a period of time.  When the blinking ends, the LEDs will indicate how the stall motors are aligned.  Every so often (about once in 20 - 40 times) one of the LEDs will end up in the wrong state.  Sometimes, by initiating movement in a track not related to the affected LED will correct the state of the affected LED.

This is the sequence I use to replicate:
1) push the individual buttons for 'Track_2', 'Track_3', and 'Mainline'.  The motors will all line up such that they are in the normal state and the LEDs will flash for 3 seconds then quit, staying in their desired state.
2) push the button for 'Track_1'.  The motors all three tracks will now align in the reverse state, the LEDs will flash, etc...
3) repeat steps 1 and 2 until you see one of the LEDs end up in the wrong state.  This may require 20 - 40 tries.  Changing the value of 'DELAY' in the code seems to affect frequency.
4) push the button for 'Track_4'.  The track four motor will align, its LED will flash and SOMETIMES, the affected LED will change to the proper state even though it was not explicitly instructed to do so.

I see this on both the Raspberry Pi 2 and 3.  I'm using Java 8 and Pi4J release 1.0

[StallMotorWithIndicator.txt](https://github.com/Pi4J/pi4j/files/437111/StallMotorWithIndicator.txt)

savageautomate, 2016-09-09:
@tomtibbetts 

I suspect this is happening due to the blink (and pulse in some cases) is handled by a background worker thread process.  Pi4J is using a scheduled executor to schedule the repeating blink job in the background.  There may be a race condition between the repeated scheduled "blink" task and the "stop blink" task.  

Here is the related blink code, let me know if you see an obvious solution.  I'm not able to dig into it right now.

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioPinImpl.java#L237

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioScheduledExecutorImpl.java#L143

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/tasks/impl/GpioBlinkTaskImpl.java

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/tasks/impl/GpioBlinkStopTaskImpl.java

Thanks, Robert

tomtibbetts, 2016-09-14:
Thank you Robert for getting back to me.

 

I did update the jar file on my Raspi 3b and haven’t seen issue so far.  Plus, it may be that for my blocking time on some of this stuff is a not easily divisible number of one blink durations I’m using.  So, for example, I may blink and LED at a rate of 250ms for a duration of 2999 ms (not 3000).

 

But, I’ll look further at the code when I get a chance.  But I do agree, in principal, that it looks to be a race condition.

 

Thanks

 

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioPinImpl.java#L237

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioScheduledExecutorImpl.java#L143

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/tasks/impl/GpioBlinkTaskImpl.java

https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/tasks/impl/GpioBlinkStopTaskImpl.java

savageautomate, 2016-12-02:
Did this issue ever resurface?  I'm closing the ticket for now, but feel free to reopen if its something we can reproduce and fix.  Thanks, Robert

closed 2016-12-02 by savageautomate

tomtibbetts, 2016-12-02:
Hi Robert,

Thanks for checking in.

 

I have not had a chance yet to look at the files you sent me.  I’m sure I could reproduce the issue next time I look at it.  I do believe it’s a race condition, but, as I mentioned, have not had a chance to pursue a fix.  I’ll re-open it if I can make it more reproduceable.

 

Thanks,

Tom Tibbetts