Pi4J/pi4j-v1 #273: DebounceGpioExample not working. Debounce functions not found.

JuhaMcEla, 2016-09-07:
I'm trying to run`DebounceGpioExample.java` but compiler don't found xxxDebounce() functions.

So compiler gives the error:
`Exception in thread "main" java.lang.RuntimeException: Uncompilabe source code - Erroneous sum type: com.pi4j.io.gpio.GpioPinDigitalInput.setDebounce at debounce.DeBounce.main (DeBounce.java:49)`

savageautomate, 2016-09-09:
I just tried this on a RPi v3 and it worked here ...

What do you get if you enter command ```pi4j --version``` on the command line?

If an older version of Pi4J is installed, it could be picking up the older version JARs in the classpath.

Thanks, Robert

JuhaMcEla, 2016-09-13:
I get `command not` found when using `pi4j --version` -command. I tried also remove downloaded `pi4j-core.jar` from dir which I add to classpath with java -cp -command and compiler gave `NoClassDefFoundError: com/pi4j/io/gpio/GpioFactory` so I think compiler didn't use any other pi4j-version?

savageautomate, 2016-12-02:
Sorry for the delay.  Were you able to get past this issue?

savageautomate, 2021-01-15:
**Closed.**  No further activity.

closed 2021-01-15 by savageautomate