Pi4J/pi4j-v1 #276: RaspiBcmPin.GPIO_13 not working even in BROADCOM_PIN_NUMBERING

salk31, 2016-09-22:
My usage is that I'm listening for change events. It is very odd as RaspiPin.GPIO_23 works but not RaspiBcmPin.GPIO_13 even though I've set broadcom pin numbering (and other usages take notice of the numbering scheme).

I'm afraid I've not characterised the problem yet but wondered if it rang any bells.

savageautomate, 2016-12-02:
Issue reproduced; looking into the problem.

savageautomate, 2016-12-02:
Fixed in latest deployed 1.2-SNAPSHOT.

closed 2016-12-02 by savageautomate

Ant1Zykl0n, 2017-01-02:
Does not seem to be fixed in 1.2-SNAPSHOT. The issue is only occurring for DigitalInputPin.  `GpioFactory.setDefaultProvider(new RaspiGpioProvider(BROADCOM_PIN_NUMBERING));` does not have any impact: RaspiPin.GPIO_21 is working in both cases and RaspiBcmPin.GPIO_05 in none. For DigitalOutputPin it is working.

savageautomate, 2017-01-09:
@Ant1Zykl0n 

I just tested it again here using BCM pins 21 (`RaspiPin.GPIO_21`) and 5 (`RaspiBcmPin.GPIO_05`) and it seems to be working properly for me.

[image]

I'm going to update my system with the latest Raspian updates just to make sure it still working in the latest builds.

Ant1Zykl0n, 2017-01-09:
I just tested it again. BCM as InputPin is not working. I'm using Snapshot 1.2. Here is the code

`public static void main(String[] args) throws InterruptedException {
        testInputBCM();
    }

    //working
    public static void testOutput() throws InterruptedException {
        GpioController gpioController = GpioFactory.getInstance();
        GpioPinDigitalOutput pin = gpioController.provisionDigitalOutputPin(RaspiPin.GPIO_21);

        pin.setShutdownOptions(true, LOW);
        pin.setState(HIGH);
        Thread.sleep(5000);
        gpioController.shutdown();
    }

    //working
    public static void testOutputBCM() throws InterruptedException {
        GpioFactory.setDefaultProvider(new RaspiGpioProvider(BROADCOM_PIN_NUMBERING));
        GpioController gpioController = GpioFactory.getInstance();
        GpioPinDigitalOutput pin = gpioController.provisionDigitalOutputPin(RaspiBcmPin.GPIO_05);

        pin.setShutdownOptions(true, LOW);
        pin.setState(HIGH);
        Thread.sleep(5000);
        gpioController.shutdown();
    }

    //working
    public static void testInput() throws InterruptedException {
        GpioController gpioController = GpioFactory.getInstance();
        GpioPinDigitalInput pin = gpioController.provisionDigitalInputPin(RaspiPin.GPIO_21);

        pin.addListener((GpioPinListenerDigital) event -> System.out.println("Input GPIO 21"));
        Thread.sleep(5000);
        gpioController.shutdown();
    }

    //not working
    public static void testInputBCM() throws InterruptedException {
        GpioFactory.setDefaultProvider(new RaspiGpioProvider(BROADCOM_PIN_NUMBERING));
        GpioController gpioController = GpioFactory.getInstance();
        GpioPinDigitalInput pin = gpioController.provisionDigitalInputPin(RaspiBcmPin.GPIO_05);

        pin.addListener((GpioPinListenerDigital) event -> System.out.println("Input GPIO BCM (05)"));
        Thread.sleep(5000);
        gpioController.shutdown();
    }`

salk31, 2017-04-26:
FWIW BCM 12, 13 and 14 working for me with SNAPSHOT-1.2 fetched today.

Many thanks!

commit 896f2d88 (2016-12-02): Fixed #276; Broadcom pin numbering impl was not exporting correct pins and thus pin events were not firing