Pi4J/pi4j-v1 #277: I2C Scan

guedressel, 2016-10-06:
I can't find a way to scan for active I²C devices through Pi4J.
As far as I know a "scan" operation just does an I²C begin transmission followed by an immediate I²C end transmission. If the transmission gets an ACK a device is available at that address.

Is there a way to do such scans?

natdan, 2016-10-07:
You'll have to do it manually - create device with specific address, try read if/when it succeeded record it and move on... I would expect the code to be something like:
`
List<Integer> validAddresses = new ArrayList<Integer>();
final I2CBus bus = I2CFactory.getInstance(I2CBus.BUS_1);
for (int i = 1; i < 128; i++) {
  try {
      I2CDevice device  = bus.getDevice(i);
      device.read(0); // Or something like that
     validAddresses.add(i);
  } catch (Exception ignore) { }
}
`

guedressel, 2016-10-07:
Don't you think reading a (random/blank) value could lead to an error?
Not an "there was no device"-error but "there was an actual reply from a device but it stated an error".
I think it could be that in this scenario devices could be shadowed by bad luck of reading the wrong register of a device.

But since I'm no expert in I²C communication I might think too complicated and your approach is good to go...

natdan, 2016-10-07:
You are right - it was just me trying to knock off something really quickly. I'll look at how to 'read without reading' from a device (address it only) and return back to you.

natdan, 2016-10-09:
I've just tested it as this seems to be working:

```
import java.util.ArrayList;
import java.util.List;

import com.pi4j.io.i2c.I2CBus;
import com.pi4j.io.i2c.I2CDevice;
import com.pi4j.io.i2c.I2CFactory;

public class Scan {

    public static void main(String[] args) throws Exception {
        List<Integer> validAddresses = new ArrayList<Integer>();
        final I2CBus bus = I2CFactory.getInstance(I2CBus.BUS_1);
        for (int i = 1; i < 128; i++) {
          try {
              I2CDevice device  = bus.getDevice(i);
              device.write((byte)0);
             validAddresses.add(i);
          } catch (Exception ignore) { }
        }

        System.out.println("Found: ---");
        for (int a : validAddresses) {
            System.out.println("Address: " + Integer.toHexString(a));
        }
        System.out.println("----------");
    }
}
```

Original detect tool output:
```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- 53 -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- 69 -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77  
```

Above Java code output:
```
Found: ---
Address: 1e
Address: 53
Address: 69
Address: 77
----------
```

natdan, 2016-10-09:
BTW let me know if you are happy with the solution so we can close the issue.

closed 2017-01-08 by natdan