Pi4J/pi4j-v1 #284: Classes in extension package should not create internal Threads

ggoerlich, 2016-10-25:
I want to use the components of the extension package in a larger environment and i dislike the internal polling threads in the classes of the extension package.

The classes in the extension package should use a scheduler service with default to use the scheduler service of the pi4j concurrent package and a interface to set the service to use.

Is there any work to do this change in the next version ? If not and you think this is a good change i can make a pull request, as i have made the requested changes for my tests.

savageautomate, 2016-12-02:
I think you are absolutely correct :-)  

It look like this was fixed in the SerialFactory for serial impl some time back, but never made its way to the extensions.  https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/serial/SerialFactory.java

Please make the pull request and we can merge the fixes.  

Thank You,
Robert

savageautomate, 2021-01-12:
**CLOSED**; all new development efforts are in Pi4J v2.x (https://v2.pi4j.com)

closed 2021-01-12 by savageautomate