Pi4J/pi4j-v1 #286: NoClassDefFoundError: com/pi4j/io/I2CFactory$UnsupportedBusNumberException

gjwo, 2016-10-28:
I am trying to run java code on a Raspberry Pi model 3, downloaded from a PC eclipse development environment to access a 9DoF device on the I2C bus using the pi4j library. I get error in the title. Full details here on stackoverflow
[http://stackoverflow.com/questions/40287070/noclassdeffounderror-com-pi4j-io-i2cfactoryunsupportedbusnumberexception?noredirect=1#comment67833403_40287070]

natdan, 2016-10-28:
Are you sure that ".:classes:/opt/pi4j/lib/'*'" brings everything from /opt/pi4j/lib? I am not aware that '*' works in the classpath. I think you should enumerate all needed jar files. Check what is in /opt/pi4j/lib and add all .jar files from there in a list (separated with ':'). 

Also, there are other alternatives running your jar. One is, for instance, making it into uberjar, onejar or similar package where all dependencies are packaged up together in one jar. That way you won't need to add -classpath argument.

Beside that, while developing on your PC (not RPi) you can use something like ([http://remotevm.abstracthorizon.org/eclipse-tutorial.html]) and it will take care of your classpath.

gjwo, 2016-10-28:
This sounds like the right sort of area to investigate, I have tried substituting for the '*' in my .pom file in eclipse see snippet below - no change to the result

```
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-antrun-plugin</artifactId>
				<version>1.8</version>
				<executions>
					<execution>
						<id>ant-copy-execute</id>
						<phase>install</phase>
						<configuration>

							<target>
								<!-- ensure the target directory exists on the Raspberry Pi -->
								<sshexec host="${pi.host}" port="${pi.port}" username="${pi.user}"
									password="${pi.password}" trust="true" failonerror="false"
									verbose="true" command="mkdir --parents ${pi.deployDirectory}" />

								<!-- copy the JAR file to the Raspberry Pi -->
								<scp file="${project.build.directory}/${project.build.finalName}.jar"
									todir="${pi.user}:${pi.password}@${pi.host}:${pi.deployDirectory}"
									port="${pi.port}" trust="true" verbose="true" failonerror="true" />

								<!-- run the JAR file on the Raspberry Pi -->
								<sshexec host="${pi.host}" port="${pi.port}" username="${pi.user}"
									password="${pi.password}" trust="true" failonerror="false"
									verbose="true"
									command="java -classpath .:classes:/opt/pi4j/lib/pi4j-core.jar:/opt/pi4j/lib/pi4j-device.jar:/opt/pi4j/lib/pi4j-gpio-extension.jar -jar ${pi.deployDirectory}/${project.build.finalName}.jar" />
							</target>

						</configuration>
						<goals>
							<goal>run</goal>
						</goals>
					</execution>
				</executions>
				<dependencies>
					<dependency>
						<groupId>org.apache.ant</groupId>
						<artifactId>ant-jsch</artifactId>
						<version>1.9.6</version>
					</dependency>
				</dependencies>
			</plugin>

```
I am still learning how to use maven properly so I wonder if I should also have an execution dependency naming pi4j?

gjwo, 2016-10-28:
I also tried putting all of the pi4J .jar files in the same directory as my jar file, this made no difference

natdan, 2016-10-28:
Check this example: https://www.mkyong.com/maven/maven-create-a-fat-jar-file-one-jar-example/
It might help you creating executable jar with all dependencies in it. Alternatively - do check what dependencies you really need with:
```
mvn dependency:list
```

and specify all jar files from them. Do check if files you are specifying in -cp (-classpath) arg are there as java won't complain if you define bogus/non-existent file (those are common mistakes I normally make).

gjwo, 2016-10-28:
I ran dependendy:list within Eclipse and got this

> [INFO] Scanning for projects...
> [INFO]                                                                         
> [INFO] ------------------------------------------------------------------------
> [INFO] Building RPITank 1.0-SNAPSHOT
> [INFO] ------------------------------------------------------------------------
> [INFO] Downloading: https://oss.sonatype.org/content/groups/public/com/pi4j/pi4j-core/1.2-SNAPSHOT/maven-metadata.xml
> [INFO] Downloaded: https://oss.sonatype.org/content/groups/public/com/pi4j/pi4j-core/1.2-SNAPSHOT/maven-metadata.xml (2 KB at 1.1 KB/sec)
> [INFO] Downloading: https://oss.sonatype.org/content/groups/public/com/pi4j/pi4j-parent/1.2-SNAPSHOT/maven-metadata.xml
> [INFO] Downloaded: https://oss.sonatype.org/content/groups/public/com/pi4j/pi4j-parent/1.2-SNAPSHOT/maven-metadata.xml (2 KB at 4.1 KB/sec)
> [INFO] 
> [INFO] --- maven-dependency-plugin:2.8:list (default-cli) @ RPITank ---
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting-api/2.0.9/maven-reporting-api-2.0.9.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting-api/2.0.9/maven-reporting-api-2.0.9.pom (2 KB at 11.4 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting/2.0.9/maven-reporting-2.0.9.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting/2.0.9/maven-reporting-2.0.9.pom (2 KB at 34.3 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/doxia/doxia-sink-api/1.0-alpha-10/doxia-sink-api-1.0-alpha-10.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/doxia/doxia-sink-api/1.0-alpha-10/doxia-sink-api-1.0-alpha-10.pom (2 KB at 32.1 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/doxia/doxia/1.0-alpha-10/doxia-1.0-alpha-10.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/doxia/doxia/1.0-alpha-10/doxia-1.0-alpha-10.pom (9 KB at 248.5 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting-api/3.0/maven-reporting-api-3.0.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting-api/3.0/maven-reporting-api-3.0.pom (3 KB at 82.8 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/shared/maven-shared-components/15/maven-shared-components-15.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/shared/maven-shared-components/15/maven-shared-components-15.pom (10 KB at 217.1 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/maven-parent/16/maven-parent-16.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/maven-parent/16/maven-parent-16.pom (23 KB at 598.1 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/apache/7/apache-7.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/apache/7/apache-7.pom (15 KB at 380.9 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting-impl/2.0.5/maven-reporting-impl-2.0.5.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/reporting/maven-reporting-impl/2.0.5/maven-reporting-impl-2.0.5.pom (5 KB at 142.2 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/shared/maven-doxia-tools/1.0.2/maven-doxia-tools-1.0.2.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/shared/maven-doxia-tools/1.0.2/maven-doxia-tools-1.0.2.pom (6 KB at 108.1 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/maven/shared/maven-shared-components/11/maven-shared-components-11.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/maven/shared/maven-shared-components/11/maven-shared-components-11.pom (9 KB at 254.5 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/commons-io/commons-io/1.4/commons-io-1.4.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/commons-io/commons-io/1.4/commons-io-1.4.pom (13 KB at 357.2 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/org/apache/commons/commons-parent/7/commons-parent-7.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/org/apache/commons/commons-parent/7/commons-parent-7.pom (18 KB at 438.1 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/commons-validator/commons-validator/1.2.0/commons-validator-1.2.0.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/commons-validator/commons-validator/1.2.0/commons-validator-1.2.0.pom (9 KB at 260.9 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/commons-beanutils/commons-beanutils/1.7.0/commons-beanutils-1.7.0.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/commons-beanutils/commons-beanutils/1.7.0/commons-beanutils-1.7.0.pom (357 B at 12.5 KB/sec)
> [INFO] Downloading: https://repo.maven.apache.org/maven2/commons-logging/commons-logging/1.0.3/commons-logging-1.0.3.pom
> [INFO] Downloaded: https://repo.maven.apache.org/maven2/commons-logging/commons-logging/1.0.3/commons-logging-1.0.3.pom (866 B at 31.3 KB/sec)
[... cut ...]
> [INFO] ------------------------------------------------------------------------
> 
Which wasn't very helpful since it is all about maven not really what my code uses from pi4J, interesting to know how to do this though. 

Is there a to check what classes are in the pi4j jars? if the missing one is there then we know that it is an execution environment linkage problem. I would prefer not to bundle pi4j in with my classes since it is installed on the RaspberryPi, and having two copies seems like a waste of space, and a less ideal solution.

natdan, 2016-10-28:
Could you, please, share rest of the plugin? It seems that you didn't set your dependencies correctly as only dependency shown here is pi4j-core and that's only for compile time... Maybe that's the reason onejar/uberjar doesn't work (if you have configured it after all).

gjwo, 2016-10-28:
OK, I have used tattletale to find the dependencies, they are
```
Class	Dependants
com.pi4j.io.gpio.GpioController	main.DriveAssemblyTest
com.pi4j.io.gpio.GpioFactory	main.DriveAssemblyTest
com.pi4j.io.gpio.GpioPinDigitalOutput	devices.motors.DCMotor, devices.motors.StepperMotor, main.DriveAssemblyTest
com.pi4j.io.gpio.Pin	devices.motors.DCMotor, main.DriveAssemblyTest
com.pi4j.io.gpio.PinState	devices.motors.DCMotor, devices.motors.StepperMotor, main.DriveAssemblyTest
com.pi4j.io.gpio.RaspiPin	main.DriveAssemblyTest
com.pi4j.io.i2c.I2CBus	devices.sensorImplementations.HMC5883L, inertialNavigation.Navigate, main.MPU9250Test
com.pi4j.io.i2c.I2CDevice	devices.I2C.Pi4jI2CDevice, devices.sensorImplementations.HMC5883L, inertialNavigation.Navigate, main.MPU9250Test
com.pi4j.io.i2c.I2CFactory	devices.sensorImplementations.HMC5883L, inertialNavigation.Navigate, main.MPU9250Test
com.pi4j.io.i2c.I2CFactory$UnsupportedBusNumberException	devices.sensorImplementations.HMC5883L, inertialNavigation.Navigate, main.MPU9250Test
com.pi4j.wiringpi.SoftPwm	devices.motors.DCMotor
```
The repository is here https://github.com/MAWoodMain/RPITank/tree/EclipseToPiDeployment

The full .pom is here

```
<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0"
         xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
         xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">

    <modelVersion>4.0.0</modelVersion>

    <groupId>me.mawood</groupId>
    <artifactId>RPITank</artifactId>
    <version>1.0-SNAPSHOT</version>
 	<name>RPITank</name>
	<description>Navigation for a model tank</description>
 
    <properties>
    	<project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>
		<maven.compiler.source>1.8</maven.compiler.source>
		<maven.compiler.target>1.8</maven.compiler.target>
		<!-- DEFAULT RASPBERRY PI PROPERTIES -->
		<pi.host>192.168.1.127</pi.host>
		<pi.port>22</pi.port>
		<pi.user>pi</pi.user>	
		<pi.password>raspberry</pi.password>
		<pi.deployDirectory>/home/pi/artifacts</pi.deployDirectory>
		<pi.main.class>main.MPU9250Test</pi.main.class>
  	</properties>
  	
    <repositories>
        <repository>
            <id>oss-snapshots-repo</id>
            <name>Sonatype OSS Maven Repository</name>
            <url>https://oss.sonatype.org/content/groups/public</url>
            <snapshots>
                <enabled>true</enabled>
                <updatePolicy>always</updatePolicy>
            </snapshots>
        </repository>
    </repositories>
    
    <dependencies>
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-core</artifactId>
            <version>1.2-SNAPSHOT</version>
        </dependency>
    </dependencies>

    <build>
        <plugins>
            <plugin>
                <groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-compiler-plugin</artifactId>
				<version>3.5.1</version>
				<configuration>
					<source>1.8</source>
					<target>1.8</target>
				</configuration>
            </plugin>

			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-jar-plugin</artifactId>
				<version>3.0.2</version>
				<configuration>
					<archive>
						<manifest>
							<addClasspath>true</addClasspath>
							<mainClass>${pi.main.class}</mainClass>
						</manifest>
					</archive>
				</configuration>
			</plugin>

			<!--This plugin will transfer the executable JAR file to the Pi and runs 
				it -->
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-antrun-plugin</artifactId>
				<version>1.8</version>
				<executions>
					<execution>
						<id>ant-copy-execute</id>
						<phase>install</phase>
						<configuration>

							<target>
								<!-- ensure the target directory exists on the Raspberry Pi -->
								<sshexec host="${pi.host}" port="${pi.port}" username="${pi.user}"
									password="${pi.password}" trust="true" failonerror="false"
									verbose="true" command="mkdir --parents ${pi.deployDirectory}" />

								<!-- copy the JAR file to the Raspberry Pi -->
								<scp file="${project.build.directory}/${project.build.finalName}.jar"
									todir="${pi.user}:${pi.password}@${pi.host}:${pi.deployDirectory}"
									port="${pi.port}" trust="true" verbose="true" failonerror="true" />

								<!-- run the JAR file on the Raspberry Pi -->
								<sshexec host="${pi.host}" port="${pi.port}" username="${pi.user}"
									password="${pi.password}" trust="true" failonerror="false"
									verbose="true"
									command="java -classpath .:classes:/opt/pi4j/lib/pi4j-core.jar:/opt/pi4j/lib/pi4j-device.jar:/opt/pi4j/lib/pi4j-gpio-extension.jar -jar ${pi.deployDirectory}/${project.build.finalName}.jar" />
							</target>

						</configuration>
						<goals>
							<goal>run</goal>
						</goals>
					</execution>
				</executions>
				<dependencies>
					<dependency>
						<groupId>org.apache.ant</groupId>
						<artifactId>ant-jsch</artifactId>
						<version>1.9.6</version>
					</dependency>
				</dependencies>
			</plugin>
        </plugins>
    </build>
</project>
```

I haven't attempted to use one jar or uberjar as per my previous comment. What I want to to is establish the link to the pi4j jars on the RPI rather than copy them and put them in a jar.

gjwo, 2016-10-28:
This is interesting - output of the manifest from my jar

```
Manifest-Version: 1.0
Built-By: GJWood
Class-Path: pi4j-core-1.2-20160727.234749-7.jar
Created-By: Apache Maven 3.3.9
Build-Jdk: 1.8.0_31
Main-Class: main.MPU9250Test

### OSGi information
Bundle-ManifestVersion: 2
Bundle-SymbolicName: devices.sensors.dataTypes
Bundle-Description: RPITank-1.0-SNAPSHOT
Bundle-Name: RPITank-1.0-SNAPSHOT
Bundle-Version: 0.0.0
Export-Package: communication,devices.I2C;uses:="com.pi4j.io.i2c",devi
 ces.driveAssembly;uses:="devices.motors",devices.motors;uses:="com.pi
 4j.io.gpio,com.pi4j.wiringpi",devices.sensorImplementations;uses:="co
 m.pi4j.io.i2c,devices.sensors.dataTypes,devices.sensors.interfaces",d
 evices.sensorImplementations.MPU9250;uses:="devices.I2C,devices.senso
 rs,devices.sensors.dataTypes",devices.sensors;uses:="devices.sensors.
 dataTypes,devices.sensors.interfaces",devices.sensors.dataTypes;uses:
 ="devices.sensors.interfaces",devices.sensors.interfaces;uses:="devic
 es.sensors.dataTypes",inertialNavigation;uses:="com.pi4j.io.i2c,devic
 es.I2C,devices.sensorImplementations.MPU9250,devices.sensors.dataType
 s,devices.sensors.interfaces",main;uses:="com.pi4j.io.gpio,com.pi4j.i
 o.i2c,devices.I2C,devices.driveAssembly,devices.motors,devices.sensor
 Implementations.MPU9250,devices.sensors.dataTypes"
Import-Package: com.pi4j.io.gpio,com.pi4j.io.i2c,com.pi4j.wiringpi,jav
 a.io,java.lang,java.net,java.util
```
Note the import packages, it would be good if these could point to the correct places!

natdan, 2016-10-28:
Unfortunately it doesn't help when you call it with java -classpath. It has some value for OSGi.
Also, I've just checked - I have following two in my pom as well:
```
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-device</artifactId>
            <version>${pi4j.version}</version>
        </dependency>

        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-native</artifactId>
            <version>${pi4j.version}</version>
            <classifier>raspberrypi-dynamic</classifier>
            <type>so</type>
        </dependency>
```
Then it seems that all you need to do is to add pi4j-core.jar, pi4j-native.jar and pi4j-device.jar to the classpath.

gjwo, 2016-10-28:
What's the second one, and where in the maven lifecycle is it?

natdan, 2016-10-29:
List them next to pi4j-core dependency. Second one is dependency to native JNI code.

gjwo, 2016-10-29:
I have added the second dependency, and looked at its .pom, this looks like a missing piece of the jigsaw, however I am still getting the same result. I suspect my classpath is wrong. I have it only as an explicit part of the java command in the execution section. Where do you define this, and what is your definition?

gjwo, 2016-10-29:
Got it!
changed the jar section to this:

```
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-jar-plugin</artifactId>
				<version>3.0.2</version>
				<configuration>
					<archive>
						<manifest>
							<addClasspath>true</addClasspath>
							<!--<classpathPrefix>${pi.pi4j.Directory}/</classpathPrefix>-->
							<mainClass>${pi.main.class}</mainClass>
						</manifest>
            			<manifestEntries>
              				<!-- Add the pi4j in runtime. -->
              				<Class-Path>${pi.pi4j.Directory}/pi4j-core.jar</Class-Path>
            				</manifestEntries>
					</archive>
				</configuration>
			</plugin>
```

gjwo, 2016-10-29:
Thanks for your help

closed 2016-10-29 by gjwo