Pi4J/pi4j-v1 #288: wiringPiSetup: mmap (GPIO) failed: Invalid argument

GrmpCerber, 2016-11-13:
Hi, I've been playing around for a few weeks with several GPIO programs (python GPIO and vanilla wiringPi) on my Pi2 and they are working as expected.
My true choice was Pi4J because I'm more fluent in java than any other languages.

Yet I was surprised / puzzled by the fact that it did not work : with pi4j-core:1.1, when calling `GpioFactory.getInstance()` my program crashes / stops with the following message `wiringPiSetup: mmap (GPIO) failed: Invalid argument`

I've repackaged it with 1.0 and 1.2-SNAPSHOT but it's always failing the same way.
I've also tried to use `/dev/gpiomem` setting the env variable `WIRINGPI_GPIOMEM=true` without success either.

do you have any ideas on how to troubleshoot this ?

Regards

GrmpCerber, 2016-11-13:
I wouldn't say it's solved but, after a few hours rebuilding the supporting library from master allowed me to go forward.
It might be linked to my distro : I'm running Alpine linux (based on Busybox) on my Pi

(PS. I ran in a few compile issue. I'll submit a couple PR later)

savageautomate, 2016-12-02:
So do you think its an incompatibility with the pi4j-native compiled .so library?   We have always (and only) compiled it under Raspian or Ubuntu (with RPi cross-compiler).  

Thanks, Robert

savageautomate, 2021-01-14:
Closed; no further activity

closed 2021-01-14 by savageautomate