Pi4J/pi4j-v1 #290: Support for ARM processor AM3703 (1GHz) ?

mik0205, 2016-11-17:
Hi Robert,

I want to use the pi4j library at a new board. It is called IGEP0030.
(please see  IGEP0030  at  http://labs.isee.biz/index.php/Category:IGEP_Processor_Boards   
in the table it is the IGEP COM ELECTRON.)

Here are some informations about the os:

Poky (Yocto Project Reference Distro) 1.7.3 
Linux igep00x0 2.6.37 #7 Fri Sep 30 12:35:41 CEST 2016 armv7l GNU/Linux
Linux version 2.6.37 (VirtualBox) (gcc version 4.7.3 (Ubuntu/Linaro 4.7.3-12ubuntu1) ) 

I used pi4j-1.1.0-SNAPSHOT.

When I start my application I received this exception:

...
INFO @ 13:20:14.918 SerialPortPi4J.SerialPortPi4J: serial=com.pi4j.io.serial.impl.SerialImpl@1b7e5df
INFO @ 13:20:14.919 SerialPortPi4J.open: SerialConfig port=/dev/ttyAMA0 bitRate=7 dataBits=2 parity=2 stop=0 flow=0
Oct 13, 2016 1:20:14 PM com.pi4j.util.NativeLibraryLoader load
**SCHWERWIEGEND: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/static/libpi4j.so]
java.lang.UnsatisfiedLinkError: /var/volatile/tmp/libpi4j4391979578648822778.so: /var/volatile/tmp/libpi4j4391979578648822778.so: internal error**
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
        at java.lang.Runtime.load0(Runtime.java:809)
        at java.lang.System.load(System.java:1086)
        at com.pi4j.util.NativeLibraryLoader.loadLibraryFromClasspath(NativeLibraryLoader.java:148)
        at com.pi4j.util.NativeLibraryLoader.load(NativeLibraryLoader.java:97)
        at com.pi4j.jni.Serial.<clinit>(Serial.java:101)
        at com.pi4j.io.serial.Parity.<clinit>(Parity.java:34)
        ...

INFO @ 13:20:15.022 SerialPortPi4J.open: port=/dev/ttyAMA0 bitRate=9600 dataBits=7 parity=2 stop=1 flow=0
EXCEP@ ============================================================
EXCEP@ Exception at: 2016-10-13 13:20:15
EXCEP@          Msg:
EXCEP@ com.pi4j.jni.Serial.open(Ljava/lang/String;IIIII)I
EXCEP@ ------------------------------------------------------------
EXCEP@ java.lang.UnsatisfiedLinkError: com.pi4j.jni.Serial.open(Ljava/lang/String;IIIII)I
EXCEP@     at com.pi4j.jni.Serial.open(Native Method)
EXCEP@     at com.pi4j.io.serial.impl.SerialImpl.open(SerialImpl.java:151)
...

What can I do? Should it work with this hardware/os?

Many thanks for your assistance.

Michael

mik0205, 2016-11-17:
Here are some informations which I got by running the SystemInfoExample:

----------------------------------------------------
PLATFORM INFO
----------------------------------------------------
Platform Name     :  Raspberry Pi
Platform ID       :  raspberrypi
----------------------------------------------------
HARDWARE INFO
----------------------------------------------------
Serial Number     :  0000000000000000
CPU Revision      :  2
CPU Architecture  :  7
CPU Part          :  0xc08
Hardware          :  IGEP0030 COM
Hardware Revision :  0000
Is Hard Float ABI :  true
Board Type        :  RaspberryPi_Unknown
----------------------------------------------------
MEMORY INFO
----------------------------------------------------
Total Memory      :  398176256
Used Memory       :  314265600
Free Memory       :  83910656
Shared Memory     :  0
Memory Buffers    :  450560
Cached Memory     :  260468736
----------------------------------------------------
OPERATING SYSTEM INFO
----------------------------------------------------
OS Name           :  Linux
OS Version        :  2.6.37
OS Architecture   :  arm
----------------------------------------------------
JAVA ENVIRONMENT INFO
----------------------------------------------------
Java Vendor       :  Oracle Corporation
Java Vendor URL   :  http://java.oracle.com/
Java Version      :  1.8.0_111
Java VM           :  Java HotSpot(TM) Embedded Client VM
Java Runtime      :  Java(TM) SE Embedded Runtime Environment
----------------------------------------------------
NETWORK INFO
----------------------------------------------------
Nameserver        :  10.66.77.2
----------------------------------------------------
CODEC INFO
----------------------------------------------------
----------------------------------------------------
CLOCK INFO
----------------------------------------------------

savageautomate, 2016-12-02:
Hi Michael,

I'm guessing that its failing to load the Pi4J pre-compiled native library for the target OS/hardware.  You could attempt to rebuild the pi4j project by compiling it from the repo; however, I think you will run into more trouble as it depends on the WiringPi project which it highly specific to the Raspberry Pi hardware.  Other supported platforms built into Pi4J do depend on WiringPi ports for those alternate platforms such as BananaPi and Odroid.   If all you need is serial communication, you may be better off looking at a different library such as the RXTX project.  You may still need to compile it for your hardware, but that's probably a smaller effort than getting a full WiringPi port for your hardware.   

Thanks, Robert

closed 2016-12-02 by savageautomate