Pi4J/pi4j-v1 #291: W1Device name contains "\n"

kisdaniel, 2016-11-19:
When I want to list the temperature sensor devices, I noticed that the end of the device name contains a new line character (#13) or in other "\n"
```
final W1Master master = new W1Master();
final List<W1Device> newDevices = master.getDevices();
final List<String> newDeviceNames = newDevices.stream().map(device -> device.getName()).collect(Collectors.toList());

```
My device name was: `28-0416811930ff\n` instead of  `28-0416811930ff`

natdan, 2016-11-19:
That's wrong. I'll have a look at it tomorrow.

natdan, 2016-11-20:
Quite right. Thanks for letting us know. Quick fix is now in and should be available in next snapshot. Alternatively just rebuild it from the source. Do test it and close the issue if you're happy with it.

savageautomate, 2016-12-01:
New 1.2-SNAPSHOT build published with this fix.

closed 2016-12-01 by savageautomate