Pi4J/pi4j-v1 #297: W1 devices getting new line to sensor name

jvalkeal, 2016-12-26:
I was passing temperature sensor name as is into json and parsing failed
```
Caused by: com.fasterxml.jackson.core.JsonParseException: Illegal unquoted character ((CTRL-CHAR, code 10)): has to be escaped using backslash to be included in string.
```
This is caused by [TmpDS18B20DeviceType.java#L107](https://github.com/Pi4J/pi4j/blob/master/pi4j-device/src/main/java/com/pi4j/component/temperature/impl/TmpDS18B20DeviceType.java#L107) reading name from a `/sys/bus/w1/devices/28-0315a24b98ff/name` as is.
```
pi@raspberrypi:~ $ hexdump -C /sys/bus/w1/devices/28-0315a24b98ff/name
00000000  32 38 2d 30 33 31 35 61  32 34 62 39 38 66 66 0a  |28-0315a24b98ff.|
00000010
```
As you can see there is a new line character at end which should be just trimmed before passing that on.

I can create PR if needed.

natdan, 2016-12-27:
I thought that was fixed. Check commit ea7cce6ce892a277b263b85f656662ffb864379c.
Did you check the latest snapshot? Or is it something else? If so, please do make a PR :)

jvalkeal, 2016-12-27:
Oh, naturally I didn't have snapshots and that commit should do the trick. Sorry for the noise!

closed 2016-12-27 by jvalkeal

natdan, 2016-12-27:
No, don't worry. It is good that people care. :)