Pi4J/pi4j-v1 #299: GpioPinListenerDigital listener example fails

wmacevoy, 2017-01-02:
I have an RPi3 (just updated) and the GpioPinListenerDigital change listener does not work.  It
is a slightly modified version of

http://pi4j.com/example/listener.html

because (1) I could not get the pin configuration to work, and (2) there was no alternate viewing of the pin state, which is now reported in a simple polling loop.

https://github.com/wmacevoy/kiss/tree/master/pi/examples/blink/bug

I can get the console output to switch from HIGH to LOW, but it never invokes the change listener.

jphhoeks, 2017-01-02:
I had the same problem. The listeners don't seem to work with the Broadcom pin numbering scheme. After switching to the default numbering, it did work.

jvalkeal, 2017-01-07:
Noticed this today as well. I was debugging `GpioInterrupt` and pin is enabled via `enablePinStateChangeCallback()` but `pinStateChangeCallback()` is never called. I'll probably try wiringpi numbering tomorrow but it's weird if it works with that as native code has support for gpio, wiringpi and physical pin locations.

jvalkeal, 2017-01-08:
Difficult to say what goes wrong, something to do with mixing pin numbering or failure to setup interrupts. WiringPi outputs some debug logs:
```
wiringPi: wiringPiSetup called
piboardRev: Hardware: Hardware	: BCM2709
piBoardId: Revision string: Revision	: a02082
piBoardId: New Way: revision is: 0x00A02082
piboardId: rev: 2, type: 8, proc: 2, mfg: 0, mem: 2, warranty: 0
wiringPi: wiringPiSetupGpio called
Setting to: 32. Current: 0x00005000
Set     to: 32. Now    : 0x00020000
```
Everything works with WiringPi numbering.

savageautomate, 2017-01-08:
@wmacevoy,

I just tried this out on my RPI 3B and it seems to be working for me.
I uninstalled any previous Pi4J instances and then downloaded the Pi4J 1.2-SNAPSHOT debian package and installed it.
http://get.pi4j.com/download/pi4j-1.2-SNAPSHOT.deb

Then I used WGET to download your sample code and then compiled and ran it.
`wget https://raw.githubusercontent.com/wmacevoy/kiss/master/pi/examples/blink/bug/src/ChangeBug1.java`

To test it, I connected and disconnected a wire between a GROUND pin (_header pin 6_) and (BCM) GPIO2 (_header pin 3_) as depicted on this diagram:
[image]

Here is an example of the output I got:
<pre>
HIGH
 --> GPIO PIN STATE CHANGE: "myButton" <GPIO 2> = LOW
LOW
LOW
 --> GPIO PIN STATE CHANGE: "myButton" <GPIO 2> = HIGH
 --> GPIO PIN STATE CHANGE: "myButton" <GPIO 2> = LOW
 --> GPIO PIN STATE CHANGE: "myButton" <GPIO 2> = HIGH
HIGH
HIGH
HIGH
 --> GPIO PIN STATE CHANGE: "myButton" <GPIO 2> = LOW
LOW
</pre>

So it seems to be working for me?  Can you please try it again with the latest build.  Is there anything else that could be different? 

Thanks, Robert

savageautomate, 2017-01-08:
@jvalkeal & @jphhoeks 

What BCM pins were you guys using?  I'll test them here just to verify. 

Thanks, Robert

joeygibson, 2017-01-09:
I wish I had checked here a few days ago. This is the same problem I was having. I originally started with Kotlin + diozero, using Pi4J as the provider. I was able to turn on an LED, and I could poll my button, but events would not fire. I then switched to Java + Pi4J 1.1, to rule out any weird Kotlin<->Java issues, but got the same results. Polling the button in a loop with `button.isHigh()` worked, so I was able to read the state of the pin, it just didn't work with events. Tonight I changed from Broadcom numbering to the Pi4J native numbering, and the events fire like they are supposed to. 

This is with an RPi 3 B (http://www.mcmelectronics.com/product/83-17300). I was using pin 17/0 for output, and pin 27/2 for my button input.

jvalkeal, 2017-01-09:
Maybe I should try snapshots as well as I've been using latest release. 

Pin I used is `BCM 17` which is `wPi 0`.

savageautomate, 2017-01-09:
@jvalkeal @joeygibson 

Definitely use the latest 1.2-SNAPSHOT build.  This was a defect that was worked on recently.  See https://github.com/Pi4J/pi4j/issues/276

In the meantime I'll go ahead and check BCM 17 and 27

Thanks, Robert

savageautomate, 2017-01-09:
@jvalkeal @joeygibson 

I just tested BCM pins 17, 27 and 2 and all listeners seem to be working for me (using latest 1.2-SNAPSHOT).    

Please note that BCM pin 2 can only be pulled UP as it has an internal constant pull up resistor hardwired on the board.  Thus you can only toggle it by connecting and disconnecting a GROUND to it.  

Thanks, Robert

jvalkeal, 2017-01-09:
Oh, cheez. From now on I'll never comment any issue until I've checked snapshots(should probably know this by now as a dev). Think my mistake was that I was running 1.1 and going though code in master and thinking, well this should work. I suppose positive thing is that I now better understand how JNI stuff is integrated :)

jphhoeks, 2017-01-09:
With the latest snapshot it is working for me, too. Thanks for the support.

jvalkeal, 2017-01-13:
Finally tried it with snapshots and it works for me as well.

savageautomate, 2017-01-13:
OK, closing this ticket for the time being ... unless someone runs into a situation where its not working with latest 1.2- SNAPSHOT build (or newer).   Thanks, Robert

closed 2017-01-13 by savageautomate

roli8200, 2018-03-01:
Sorry. DOES ALREADY NOT WORK ON RASPEBRRY PI 3 AND 1.2 Snapshot

Works:
#!/bin/bash

echo "20" > /sys/class/gpio/export
echo "in" > /sys/class/gpio/gpio20/direction

watch -n 1 cat /sys/class/gpio/gpio20/value

echo "20" > /sys/class/gpio/unexport

DOES NOT WORK:

************************************************************
************************************************************

                  <-- The Pi4J Project -->
                     GPIO Listen Example

************************************************************
************************************************************

------------------------------
|    PRESS CTRL-C TO EXIT    |
------------------------------

 ... Successfully provisioned [GPIO 20] with PULL resistance = [DOWN]

---------------------------------------------------
|  Please complete the [GPIO 20] circuit and see  |
|  the listener feedback here in the console.     |
---------------------------------------------------

^C
************************************************************
                           GOODBYE
************************************************************