Pi4J/pi4j-v1 #300: GPIO State Listener Example does not work (anymore)

yglodt, 2017-01-21:
I have a Pi3 and I am trying out the example http://pi4j.com/example/listener.html which is a GPIO state listener.

My issue is the following:

When I press (close) the contact, I get one HIGH event.

When I release (open) the contact, I get LOW/HIGH-events permanently.

I also tried to debounce() the events: myButton.setDebounce(1000);

It really debounces the events. It means, with the contact opened I get only one (in fact two, it's one LOW immediately followed by a HIGH) per second.

I would like to get a single event only on state changes.

Is a code-change necessary with the Pi3 or do I miss something?

I am aware of issue 121 https://github.com/Pi4J/pi4j/issues/121, but I believe my case is different because a) it refers to an "official" code example which does not work and b) because I do not get fake readings but more something like an oscillating behaviour.

p.s.
I found here someone has the same problem: https://groups.google.com/d/msg/pi4j/hdVvEBCpRZc/HOdWFcSyHpIJ

yglodt, 2017-01-22:
I just tried the example on a Pi 2 and a Pi Zero, and there it works as expected. No oscillations, no issues. I will investigate further if the issue could be my hardware (I have a second Pi 3 which is used in prod but I might me able to try out the Listener example on it)

jvalkeal, 2017-01-29:
@yglodt are you on snapshots or latest release? see #299

yglodt, 2017-01-29:
@jvalkeal I have tried with 1.0 and 1.1.
I found the example was working fine on a Pi Zero and a Model B+.
On my Pi3 it did not work.

jvalkeal, 2017-01-29:
Ah sorry I was hoping it might have been caused by some bugs which are fixed in snapshots. Having said that it's worth to try `1.2-SNAPSHOT` as you didn't mention you tried it.

savageautomate, 2017-01-29:
Yes, please try latest 1.2-SNAPSHOT build.  

It sounds like the GPIO pin you are using is not pulled down to ground.  Pulling the pin to a ground gives it a bias and stabilizes the pin state to the LOW state when "pulled down" and the contacts on your switching circuit are OPEN (not connected).  

The example code does attempt to pull down the pin using the internal pull resistance.  Which pin number are you testing with?  There are a couples of pins that are permanently pulled up (via physical circuit) and cannot be pulled down in software.

yglodt, 2017-03-27:
I have been giving the Example another try in a new attempt and I can not reproduce the "bug" anymore. Maybe it was related to the fact that I enabled w1-communication in the kernel, which I disabled now. I also upgraded to the latest kernel and wiringpi packages, that could also be related...

closed 2017-03-27 by yglodt