Pi4J/pi4j-v1 #301: RejectedExecutionException when using both Serial and GpioController instance with release 1.1

fguiet, 2017-01-24:
Environment:
Raspberry 3
Pi4J 1.1
Oracle Java 8

Hi,

I am using Pi4J library in a personal automation project (see https://github.com/fguiet/AutomationServer if interested).
In this project, I collect electrical information through raspberry serial gpio (GPIO #15 and #16). I also change some gpio state (from high to low or from low to high). 
In order to do that, I use Serial instance and GpioController instance at the same time (Serial collects information all the time in a separate thread). 
Everything works great with Pi4J 1.0 but as soon as I try Pi4J 1.1 I got the following error in my log: 

thread "Thread-305"
java.util.concurrent.RejectedExecutionException: Task
com.pi4j.io.serial.tasks.SerialDataEventDispatchTaskImpl@736daa
rejected from
java.util.concurrent.ThreadPoolExecutor@1214351[Terminated, pool
 size = 0, active threads = 0, queued tasks = 0, completed tasks = 4]

I am happy with 1.0 but I wanted to mention this problem. Maybe it can help someone else.

Seeya
Fred

Tutankhamon, 2017-04-10:
Same problem here, but I did not try with Pi4J 1.0. Any clue?

fguiet, 2017-07-14:
Hey again,

Event if Pi4J 1.0 was running, I had to update to Pi4J 1.2-SNAPSHOT because of a recent Raspberry kernel update. Otherwise WiringPi was throwing an error like :

 "Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY."

So I solved this issue in my Java code by removing one line : "gpio.shutdown();"

Hope this help !

Fred

closed 2017-07-14 by fguiet