Pi4J/pi4j-v1 #302: pi4j not sending data to Arduino till serial monitor on pi is opened

chaudharipa, 2017-01-29:
I connected Arduino Uno over USB to rpi3 with Jessi. When I run SerialExample, it just blinks indicator LED on Arduino (same as when I open serial monitor), but does not send any data to Arduino (I checked it by configuring another arduino pin as Rx-Tx and printing O/P input on it).
Now I installed Arduino IDE on RPi, opened it and opened serial monitor. I also reset Ardunio after this. Now when I run SerialExample in terminal window, it started sending data to Arduino.
Any way I can configure RPi to directly take control of serial instead of having to open up Arduino Serial monitor?
ps: I already done config changes in boot config and also changed settings in raspi config.

iqnev, 2017-02-10:
Could you show  us your code?

chaudharipa, 2017-02-11:
Thanks. Apologies for delay in responding.

I'm using Arduino Uno, Raspberry pi 3 - connected through USB cable
Additionally my Arduino is also connected to an NRF radio module (hence the NRF library calls in below Arduino code).

Code on Arduino:
`#include <SPI.h> //Call SPI library so you can communicate with the nRF24L01+
#include <nRF24L01.h> //nRF2401 libarary found at https://github.com/tmrh20/RF24/
#include <RF24.h> //nRF2401 libarary found at https://github.com/tmrh20/RF24/

const int pinCE = 9; //This pin is used to set the nRF24 to standby (0) or active mode (1)
const int pinCSN = 10; //This pin is used to tell the nRF24 whether the SPI communication is a command or message to send out
RF24 wirelessSPI(pinCE, pinCSN); // Create your nRF24 object or wireless SPI connection
const uint64_t pAddress = 0xB00B1E6000LL;   // Radio pipe addresses for the 2 nodes to communicate.

void setup()  
{
  Serial.begin(9600);   //start serial to communicate process
  wirelessSPI.begin();            //Start the nRF24 module
  wirelessSPI.setAutoAck(1);                    // Ensure autoACK is enabled so rec sends ack packet to let you know it got the transmit packet payload
  wirelessSPI.setRetries(5,15);                 // Sets up retries and timing for packets that were not ack'd, current settings: smallest time between retries, max no. of retries
  wirelessSPI.openWritingPipe(pAddress);        // pipe address that we will communicate over, must be the same for each nRF24 module
  wirelessSPI.stopListening();
}

void loop()  
{
   int inputCommand = -1;
   if(Serial.available()) {
    char chr;
    chr = Serial.read();

    String str = "";
    str += (char) chr;
    if(isDigit(chr)) {
      inputCommand = str.toInt();
    }
    Serial.print("Command Received:");
    Serial.println(inputCommand);
   }

   if(inputCommand > -1) {   
     int t = inputCommand;
     Serial.print("Command Received1:");
    Serial.println(t);
     if (!wirelessSPI.write(&t, 4 )){  //if the send fails let the user know over serial monitor
         Serial.println("packet delivery failed");      
     } else {
        Serial.print("packet delivery Successful. Sent:");
        Serial.println(t);
     }
   }
    //delay(1000);
}`
Java Code on Raspberry:
`import com.pi4j.wiringpi.Serial;

public class ArduinoConnectionUtil {

	private static int demo1Command = 1;
	
        public static String Demo1() {

		int serialPort = Serial.serialOpen("/dev/ttyUSB0", 9600);

		if (serialPort == -1) {
			System.out.println("Serial Port failed/not connected");
			return "Serial Port failed/not connected";
		} else {
			System.out.println("Serial Port Initialized");
		}

		Serial.serialPuts(serialPort, (demo1Command + ""));

		while (Serial.serialDataAvail(serialPort) > 0) {
			char inChar = (char) Serial.serialGetchar(serialPort);
			System.out.print(inChar);
		}

		Serial.serialClose(serialPort);

		return "Demo mode 1";
	}
}`

savageautomate, 2021-02-15:
No further activity; issue closed.

closed 2021-02-15 by savageautomate