Pi4J/pi4j-v1 #303: I2C BUS choices omit half the range of the PI's PCA9548 /dev/i2c-# range

TomOke, 2017-02-07:
With the PI 3's reduction of accessible I2C buses to just i2c-1 work was done to support a pca954x i2c multiplexor. Work was done by 6by9 with a description found through: 
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=141517

With the addition of the dts overlay, and the presence of a pca954x chip (up to 8 channels with the 9548) one can then get:
/dev/i2c-1, i2c-3, i2c-4, i2c-5, i2c-6, i2c-7, i2c-8, i2c-9 and i2c-10, where the driver handles the Multiplexor's selection mask as buses are addressed (the mux is on the physical i2c-1 bus).

However the Raspberry Pi implementation:
 pi4j/pi4j-core/src/main/java/com/pi4j/io/i2c/impl/I2CFactoryProviderRaspberryPi.java

public class I2CFactoryProviderRaspberryPi extends I2CProviderImpl {

    @Override
    protected String getFilenameForBusnumber(int busNumber) throws UnsupportedBusNumberException {
        if ((busNumber < 0) || (busNumber > 3 && busNumber < 10) || (busNumber > 17)) {
            throw new UnsupportedBusNumberException();
        }

        return "/dev/i2c-" + busNumber;
    }
}

Directly takes the supplied BUS number and generates a /dev/i2c-# file name, but specifically only accepts the range 0-4, 10-17, which misses half the mux's range. The constants in:
 pi4j/pi4j-core/src/main/java/com/pi4j/io/i2c/I2CBus.java
mirror this limit, omitting 5, 6, 7, 8, 9.

Since the code will already correctly detect and report an error with a non-existant i2c dev path it should be safe to simply drop the " || (busNumber > 3 && busNumber < 10)" check and accept the full range. Adding the additional BUS_# constants would be nice, but since they directly devolve to a number one could get by without that.

natdan, 2017-02-07:
You are right. I think we should allow those bus numbers as well. Maybe even just remove restriction to everything bigger or equal to 0. I'll remove that restriction now from I2CFactoryProviderRaspberryPi and add missing bus numbers to I2CBus.java.

TomOke, 2017-02-07:
Thank you very much. 

Tom

natdan, 2017-02-07:
I've pushed changes to master. Check them out and if you are happy close the issue.

TomOke, 2017-02-07:
Worked fine. Had to manually bring in the snapshots for static and dynamic .so for bananapi, bananapro, orangepi, and odroid. Then after I used my oracle jdk8 we got a good build and I dpkg'd the .deb and was able to access all my lines on the pca9548.

Tnaks - Tom

closed 2017-02-07 by TomOke

savageautomate, 2017-02-08:
FYI, I just deployed a new build of 1.2-SNAPSHOT available for download now.