Pi4J/pi4j-v1 #310: InternalError with LinuxFile

jvalkeal, 2017-02-12:
Ok, looks like something got broken in master possibly with https://github.com/Pi4J/pi4j/commit/65e541bc393c45db1108b60c0154729bbcc5ef0e#diff-e64dac8dc75e58e99bd691f019538a6b or something else with related changes with a `LinuxFile` because I'm now getting
```
java.lang.InternalError: null
	at com.pi4j.io.file.LinuxFile.<clinit>(LinuxFile.java:87) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.io.i2c.impl.I2CBusImpl.open(I2CBusImpl.java:123) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.io.i2c.impl.I2CProviderImpl.getBus(I2CProviderImpl.java:48) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.io.i2c.I2CFactory.getInstance(I2CFactory.java:76) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.component.lcd.impl.I2CLcdDisplay.<init>(I2CLcdDisplay.java:131) ~[pi4j-device-1.2-SNAPSHOT.jar!/:na]
```

Just a head up as I didn't have time to investigate further. This `LinuxFile` looks to be a new stuff.

colinrgodsey, 2017-02-13:
@jvalkeal what JVM and version are you using? The calls that are failing are reflection calls to an internal API, which i fear might not exist in the version you're using.

jvalkeal, 2017-02-13:
```
$ java -version
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)
```
This is the arm version but code is same for all later versions afaik.

jvalkeal, 2017-02-13:
I quickly went through jdk7/jdk8/jdk9 sources and none have that `cleaner` field in `java.nio.Buffer`.

colinrgodsey, 2017-02-13:
@jvalkeal you know, you're absolutely right about that one. im very confused now how I got this build running for test at all. its starting to look like i leaked some untested commit into that last release branch. will get a fix branch up soon

colinrgodsey, 2017-02-15:
@jvalkeal I just pushed a branch with the fix here: https://github.com/colinrgodsey/pi4j

colinrgodsey, 2017-02-16:
closing with fix in #314

closed 2017-02-16 by colinrgodsey