Pi4J/pi4j-v1 #311: Getting Input/Output IOException on the underlying RandomAccess file on I2C write transfers

TomOke, 2017-02-12:
I frequently get Input/Output IOExceptions in trying to do multi-byte output to a slave. In this case I am sending an SMB block of data (size byte, followed by size# of bytes).

This can result in the I/O error, though not always. I have placed the call to write in a retry loop (up to three tries) with a delay of 100 milliseconds between) to see if things are different from the retries, and scoped it with a TotalPhase Beaqle I2C Analyzer. This shows the retries in the attached image, where the write to device 77, register 4 produces three tries:
[image]

It may be interesting to note that the amount of data will vary for each try, which is not unexpected when an exception is signaled, but a failed transfer invariably ends in the sequence of bytes "E0 01".

An example of a trace is:
java.io.IOException: Input/output error
    at java.io.RandomAccessFile.writeBytes(Native Method)
    at java.io.RandomAccessFile.write(RandomAccessFile.java:512)
    at com.pi4j.io.i2c.impl.I2CBusImpl.lambda$writeBytes$7(I2CBusImpl.java:201)
    at com.pi4j.io.i2c.impl.I2CBusImpl.runBusLockedDeviceAction(I2CBusImpl.java:245)
    at com.pi4j.io.i2c.impl.I2CBusImpl.writeBytes(I2CBusImpl.java:193)
    at com.pi4j.io.i2c.impl.I2CDeviceImpl.write(I2CDeviceImpl.java:145)
    at com.quirklogic.utils.I2C.I2C_Processor.writeRegisterBlock(I2C_Processor.java:128)
    at com.quirklogic.bmc.api.BMC_Api.setSerialNumber(BMC_Api.java:283)

Without the delay between retries it typically fails all retries, so this might indicate some interlocking issue.

Tom

TomOke, 2017-02-13:
I added a global lock on I2C I/O and we seem to be much better. Have still seen a single error that I will investigate tomorrow, but the general I/O seems pretty solid. I am using a I2C multiplexer chip so it is possible tat we are colliding since Pi4J won't know to interlock the multiplexer channels as they bottleneck into the single physical channel.

colinrgodsey, 2017-02-13:
hmm this is very odd indeed. the I2CBusImpl should provide locks at the bus level. I suppose with the mux though, the locks would be for each sub-bus with none around the main bus. 

As far as I've seen, the I2C kernel module should maintain internal locks around individual operations to a bus (and PI4J has an extra layer in the I2CBusImpl). 

Are you using one of the linux kernel i2c mux drivers, or did you do the bus-selection implementation in user land?

TomOke, 2017-02-14:
I am using the I2C mux driver in the kernel, as per the closed issue #303.
Looking at the mux source (https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/i2c/muxes/i2c-mux-pca954x.c) seems to suggest it does not do any locking, but I have only skimmed this file alone and not done a thorough look.

colinrgodsey, 2017-02-15:
Looking around at the kernel code, it seems this issue is fixed in the stable linux branch. 

RPI: https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/i2c/i2c-mux.c
Linux: https://github.com/torvalds/linux/blob/master/drivers/i2c/i2c-mux.c

Looks like it was added as part of a patch set: https://github.com/torvalds/linux/commit/fa96f0cb9b37a1a296d18f7bde63b0910852303e

So it does definitely appear that there's a greater issue with locking around the parent for I2C muxes in linux 4.4.x.

I'll maybe look into cherry-picking those changes this weekend. Chances are the locking might be part of a greater change to the i2c infrastructure, so it might be hard/impossible. Luck has it, I'll be receiving a prototype board in a few days with a pca9546 on board, so should be able to test the kernel out if I come up with anything.

colinrgodsey, 2017-02-16:
labeling 'wontfix' for now, as it does seem to be a weakness in the 4.4 kernel, and there's not really any good mechanism to use in pi4j for cross-bus locking. 

Although, we should consider adding this to the known issues.

colinrgodsey, 2017-02-23:
@TomOke I'm going to close this for now, I'll be working on a kernel patch to provide locking this weekend. I'm not sure if I'll be able to pull off the full port of the i2c-mux locking infrastructure in the newer kernels, but should at least be able to throw some locks in the device driver.

closed 2017-02-23 by colinrgodsey

colinrgodsey, 2017-02-23:
confirmed the locking is actually now in-place in the RPi 4.9.x kernel builds 👍