Pi4J/pi4j-v1 #315: Support for ASUS Tinker

steveohara, 2017-02-20:
Hi Robert,

Another one for you!
https://www.asus.com/uk/Single-board-Computer/TINKER-BOARD/

Thanks,
Steve

kloknibor, 2017-06-15:
any news on this?

bootsy52, 2019-06-25:
Here is a patch to fully support the TinkerBoard against pi4j-core current (Version 1.2)
I used the code from the links above and added the rest necessary to pi4j-core. I tested it on my TinkerBoard S with Armbian and it works.
 
And a ready pi4j-core.jar (including the libp4i.so) for immediate use.

[pi4j-core.zip](https://github.com/Pi4J/pi4j/files/3323010/pi4j-core.zip)
[pi4j-tinkerboard-patch.zip](https://github.com/Pi4J/pi4j/files/3323009/pi4j-tinkerboard-patch.zip)

bootsy52, 2019-06-25:
And the additions to pi4j-native to build libp4j.so for the tinkerboard
[pi4j-native.zip](https://github.com/Pi4J/pi4j/files/3323036/pi4j-native.zip)

eitch, 2019-07-17:
@bootsy52 i think a pull request would make much more sense as then it is easier to see the changes you made and merge them, if so desired, into the project

bootsy52, 2019-07-23:
@eitch Currently it seems that in 1.4 the support for all other boards has been dropped, so I'm not sure how much worth it is. Plus I have not running anything on github yet, that's why i thought i contribute the patch. If i got time I can try to make pull request.

savageautomate, 2021-01-15:
**Closed.**  

As of Pi4J v2.x, board/platform expansion will provided via pluggable modules and support for specific platforms beyond the RPI is outside of the scope of the Pi4J project.   We had to make this change because supporting and testing 'N' number of board variants became impossible for a small volunteer team to keep up with.  Moving forward, Pi4J will provide the core framework and infrastructure and if someone wants to create _drivers_ for alternate platforms/boards, they will be able to do that and maintain it separately from the Pi4J codebase.  

Pi4J v2 also does not mandate/assume an implementation like WiringPi.

See: https://v2.pi4j.com/documentation/architecture

closed 2021-01-15 by savageautomate