Pi4J/pi4j-v1 #318: I2C fails with jdk9 (early access build)

jvalkeal, 2017-02-24:
@colinrgodsey just a heads up, as I predicted things go south with jdk9. We probably need to poke around what is the actual error.
```
java.lang.ExceptionInInitializerError: null
	at com.pi4j.io.i2c.impl.I2CBusImpl.open(I2CBusImpl.java:123) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.io.i2c.impl.I2CProviderImpl.getBus(I2CProviderImpl.java:48) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.io.i2c.I2CFactory.getInstance(I2CFactory.java:76) ~[pi4j-core-1.2-SNAPSHOT.jar!/:na]
	at com.pi4j.component.lcd.impl.I2CLcdDisplay.<init>(I2CLcdDisplay.java:131) ~[pi4j-device-1.2-SNAPSHOT.jar!/:na]
```

jdk version I tried
```
pi@raspberrypi:~/work $ /home/pi/work/jdk9/bin/java -version
java version "9-ea"
Java(TM) SE Runtime Environment (build 9-ea+157)
Java HotSpot(TM) Client VM (build 9-ea+157, mixed mode)
```

colinrgodsey, 2017-03-08:
Hmm, yea looks like we'll need to come up with a more compatible solution. I'll be publishing an update at some time in the near future for V4L2 support, I'll look into LinuxFile options that don't use reflection. Was mostly trying to avoid as much custom JNI as possible, but it might be inevitable.

savageautomate, 2021-02-15:
I2C should be working in latest versions Pi4J v1.3 (JDK8) and v1.4 (JDK10)

closed 2021-02-15 by savageautomate