Pi4J/pi4j-v1 #319: Hardware : BCM2835 error on RaspberryPi B+

Azn9, 2017-02-24:
Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

I use a RaspberryPi B+.

dyercode, 2017-02-25:
Also seeing this on a Zero and a B+.  It was previously working on the B+.

```
Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.
user@system ~> gpio -v
gpio version: 2.38
Copyright (c) 2012-2017 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi Zero, Revision: 03, Memory: 512MB, Maker: Sony
  * Device tree is enabled.
  *--> Raspberry Pi Zero Rev 1.3
  * This Raspberry Pi supports user-level GPIO access.
```

savageautomate, 2017-02-25:
It looks like this was a change as of Kernel v4.8.  WiringPi did add support for this change in December. See this commit:  https://github.com/Pi4J/wiringPi/commit/b1dfc186efe327aa1d59de43ef631a2fa24e7c95#diff-672071e4cd5d0146a80fb586c87c38e1R723

Can you please try Pi4J Version 1.2-SNAPSHOT to see if its working on your systems?
http://pi4j.com/download.html

The SNAPSHOT builds are compiled against the latest WiringPi builds. 

Maybe we need to go back and produce a new release of v1.1 source base compiled with the latest WiringPi?

Thanks, Robert

savageautomate, 2017-02-25:
One other option if you need to use the existing v1.1 release of Pi4J is to use the dynamic linking option where Pi4J is dynamically linked to WiringPi rather than the default static linking.

Build and install the latest WiringPi on your system.
http://wiringpi.com/download-and-install/

Now when starting your Java app, you need to set the system property of "`pi4j.linking=dynamic`"

Here is an example command line where I'm defining the system property via command line:
`sudo java -Dpi4j.debug -Dpi4j.linking=dynamic -classpath .:classes:/opt/pi4j/lib/'*' BlinkGpioExample`

Note that I also define the system property of `pi4j.debug` so that I can verify that Pi4J comes up and uses the dynamic linking. 

My output looks something like this:

```
^Cpi@rpi3b:~/pi4j-dev $ sudo java -Dpi4j.debug -Dpi4j.linking=dynamic -classpath .:classes:/opt/pi4j/lib/'*' BlinkGpioExample
<--Pi4J--> GPIO Blink Example ... started.
Feb 25, 2017 3:41:26 PM com.pi4j.util.NativeLibraryLoader load
FINE: Attempting to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]
Feb 25, 2017 3:41:26 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] has already been loaded; no need to load again.
Feb 25, 2017 3:41:26 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] has already been loaded; no need to load again.
Feb 25, 2017 3:41:26 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] has already been loaded; no need to load again.
Feb 25, 2017 3:41:26 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] loaded successfully using embedded resource file: [/lib/raspberrypi/dynamic/libpi4j.so]
Feb 25, 2017 3:41:26 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] has already been loaded; no need to load again.
 ... the LED will continue blinking until the program is terminated.
 ... PRESS <CTRL-C> TO STOP THE PROGRAM.
```

Thanks, Robert

dyercode, 2017-02-25:
Both the snapshot version and the dynamic linking on 1.1 worked for me.

savageautomate, 2017-02-25:
@dyercode ,

Thanks for the update and glad to hear both alternate options are working!

patriciowallace, 2017-03-01:
Hi, I use the static option because is more portable for me
the static option with the snapshot 1.2 is not working.

Any other sugestion ?

Regards
Patricio

savageautomate, 2017-03-01:
@patriciowallace 

Static + 1.2-SNAPSHOT should be working!

Please make sure to download the latest 1.2-SNAPSHOT build here:
http://pi4j.com/download.html

Thanks, Robert

patriciowallace, 2017-03-01:
I use the core.jar and device.jar if it helps

Regards
Patricio

patriciowallace, 2017-03-02:
any answer please

savageautomate, 2017-03-02:
I have not had a chance to test it here locally yet to confirm.  Which Pi model are you working with?

Azn9, 2017-03-03:
Hi I've tested pi4j 1.2 snapshot, and it works fine on raspi B+!

jesperldk, 2017-04-18:
1.2 snapshot works on raspi 3. 
When will we have a proper release?
And perhaps you could update the home page right away - I am sure this is confusing many new users as my self.
Thanks! :-)

zzApotheosis, 2017-04-30:
## Using
Raspberry Pi 3 - Model B
Pi4J 1.1

## Problem
Looks like I'm getting the same problem now. It literally just started happening after a Raspberry Pi reboot. I'm using the Raspberry Pi 3 - Model B.

The only difference I can think of between reboots is that I disabled some interfaces in Raspbery Pi Configuration (disabled Camera, I2C, and Serial). I don't really see how disabling those interfaces would mess up my hardware version.

NorthFred, 2017-05-03:
Problem still persists on Raspberry Pi 3 (B) with the Pi4J 1.2 Snapshot.
Fresh image:  Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26
Hardware is still detected as BCM2835.

vinny4, 2017-05-09:
After updating my rpi couple of days ago - i have this hardware version now 
Hardware : BCM2835
Revision : a22082
Serial : 0000000056f0096a
with kernel version Linux XXXRPi 4.9.25-v7+ #995 SMP Thu May 4 15:52:25 BST 2017 armv7l GNU/Linux.

But the problem still persists.
its really annoying now :(

NorthFred, 2017-05-10:
The only reasonable solution so far is to downgrade the kernel to  v.4.4.49.

kisdaniel, 2017-05-19:
I also had a same issue with raspberry pi 3, pi4j 1.1, and 4.9.24-v7 linux kernel version.
```
gpio -v
gpio version: 2.44
Copyright (c) 2012-2017 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

uname -a
Linux garazs 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux

```

After upgrading to pi4j 1.2-SNAPSHOT this problem disappeared.

When will be available 1.2 as a release?

RGeorgy, 2017-05-26:
Thanks to the team.
I have done some simple tests with pi4j snapshot 1.2 - For me it seems to work so far!
I am using Raspberrry Pi3B  Upgraded and updated 26th of May 2017.
Simply exchanged the main pi4j...jar files in my library.
Best regards

duckie96, 2017-05-28:
Hi, noob needing some help please.
When I issue the gpio command, I get the error as above.

pi@piname:~ $ gpio -g write 4 1
Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

It is a rea pi. 

pi@study:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 100000e
Serial          : 00000000765f50df

I upgraded pi4j, as mentioned above, no effect.

pi@study:~ $ dpkg -l | grep pi4j
ii  pi4j                                                        1.2~SNAPSHOT                              all          Java library for IO access on the Raspberry Pi platform

I am using a straightforward php page, to issue the gpio command.  

Not sure how to go forward with this. 
Please any advice would be appreciated.

duckie96, 2017-05-28:
Ahh got it, re installed wiringPi. . All sorted now.

manorius, 2017-05-29:
The only thing that worked for me is the solution provided by neuralassembly here: https://github.com/WiringPi/WiringPi-Python/issues/47

camueller, 2017-06-05:
Using a fresh install 2017-04-10-raspbian-jessie-lite.img followed by "apt-get update" and "apt-get upgrade"  all I have to do is switching to pi4j 1.2-SNAPSHOT. Without pi4j 1.1 I get the "Unable to determine hardware version. I see: Hardware : BCM2835" error.
Please consider a release of pi4j 1.2 soon!

pi@raspberrypi:/app $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000004190e153

NeededAnUnusedName, 2017-07-02:
I wanted to make my raspberry projects in java, so I could easiily use the GUI on every OS... Now after I spent hours of research (lib, IDE, GUI designer) I will switch back to python because of this error (which took even more hours of my lifetime). Everything works fine with python, so to all of you that just want to concentrate on a raspberry pi project rather then fighting with some stupid lib I suggest you give it a shot. 
#made_me_angry_on_weekend

thhart, 2017-12-07:
For all who probably missed it: "pi4j.linking=dynamic" is very important to specify in the Java call, see posts above. For me this fixed all problems.
Further more make sure wiringpi is really the latest installed on your Pi: 2.44

horschi, 2017-12-07:
"pi4j.linking=dynamic" worked for me on my Raspberry Pi Zero. Thanks!

DRChess, 2017-12-08:
pi4j.linking=dynamic on a command doesn't work, it is not a command.
Execution is not possible and it makes people fell left out. What did you really do?

?? Did you open some program to add it to some line of code????
?? Is there an update order that needs done??
?? Did you use from a terminal screen and add something??
?? What did you do ??
Please help by helping. 
This would be a better example:
 From the terminal screen use.
sudo java -Dpi4j.linking=dynamic

or should it be done in a specific location.
Will this need done every time I reboot or update, is it permanent? 

Remember the person doing pi4j lost interest and most of the programing modules is with pi NOOBS without a release of snapshot 1.2.

zzApotheosis, 2017-12-08:
The full command would be:
sudo java -pi4j.linking=dynamic -jar <your jar file>

Or:
sudo java -Dpi4j.linking=dynamic -jar <your jar file>

Markusmoo, 2018-01-07:
Hi, I too have been having a problem with the mentioned error.

I updated my pi4j to the 1.2 Snapshot.

This seems to have fixed my issue for my Raspberry Pi 3 Model B.
However the issue still occurs on my Raspberry Pi Zero W. :(

I have also tried the following:
sudo java -Dpi4j.linking=dynamic -jar
with the snapshot 1.2 and still receive the error.

Regards,
Markus

germanncorbetta, 2018-01-07:
Raspberry Pi 3 updated on January 2018.

Added pi4j 1.2 Snapchot (ZIP) liberary to my project worked ok.

Just added core.jar and devices.jar to the project.

THANKS!

mesutpiskin, 2018-01-27:
`-Dpi4j.linking=dynamic` working for me

sonic-1, 2018-02-19:
I have the same issue, nothing above helped but I think it has a different solution. 
 I ran this:
>
> ./pifm sampleaudio.wav fm.ft
>
> which worked.
> Then I ran:
> sudo ./rpitx -m RF -i fm.ft -f 100000 -l
>
> and got:
>
> -expecting BCM2708 or BCM2709. Please report this to projects@drogon.net
>
> rpitx Version 0.1 compiled Feb 16 2018 (F5OEO Evariste)
> running on pi@reaspberrypi . I see: Hardware : BCM2835

Gordon at drogon replied to me and said that rpitx must have a static assignment for gpio, which won't work with his code. He also said that his code won't support PWM which is probably required for riptix. 

I emailed the author of riptix about this, and am waiting for a response. 
If anyone knows what is going on, I would greatly appreciate your input.

sonic-1, 2018-02-19:
For the record, I just fixed the problem, so if someone else has the same issue, this will  help them. Just download the latest version or rptix and use that instead of what comes with qtcsdr.

SamOrozco, 2018-02-25:
Using the snapshot fixed this issue for me.

ricardojlrufino, 2018-03-05:
I get a filesystem error (EXT4-fs error) and hangs raspberry when i use PWM and SPI 
Ref.: https://github.com/mattjlewis/diozero/issues/32

eitch, 2018-03-05:
Is your app running as root? Sadly PWM needs root access otherwise it hangs.

ricardojlrufino, 2018-03-09:
> Is your app running as root? Sadly PWM needs root access otherwise it hangs.

Running as ROOT worked... thanks

DRChess, 2018-03-18:
I have had some conversations with Jacobs of JMRI and there seems a lot of forgotten history. Does not remember telling the community that you must be logged into as root for the Raspberry Pi gpio to work. So now you do not log in as root.

Currently since JMRI that came out in 2017 major changes to the Pi's OS has also changed. The program for accessing the gpio is being distributed by the Pi foundation. This has happen because Pi4j is dead as can be. No one is supporting it and JMRI needs it badly. So how to make it work. It seems the OS is on your side. 

Log in as normal.

Install new JMRI

don't bother with being root (no pi4j not needed)

add the patch to default setting as per JMRI. (this tells the java to search elsewhere for gpio dynamically)

make sure if a launcher icon is used that sudo is missing. If it finds sudo it will not use the patch.

When JMRI opens it will use the patch setting and the OS will find Gordon's closed minded open source software, version 2.44, that works on he Pi with BCM2835 coded even if the core is BCM2837. and gpio is restored to JMRI.

If this seems like a little bitterness, It took longer than I like and to get a response and to find people from Egypt living on the river of denial filled with political bias was not what I wanted to find. The Pi is still a year and a half from 2g of RAM and the Orange, Pine, and a few other have it. It will mean much more when Panel Pro runs.

By the way. DCC++ is working with digital IO (direct turnout control) and is now a $40 full DCC system with JMRI. I spent over $400.00 for mine without turnout control or sensors. The JMRI found a bug that some one reported as fixed and it wasn't.

DRChess, 2018-03-19:
The snapshot version is a developmental release and many people prefer not to bind software to it because of stability.

Since pi4j version 1.2 is not being released the question is "Has Raspberry Foundation given enough to the community to give you those missing functions?" GPIO is included with the new distribution and Java static makes software look inside of itself that causes an old GPIO function to be found that results in the error. Switching to dynamic Java looks to the bin folder where if finds the new GPIO that does work. For this reason it is sometimes not nessacery to install pi4j.

New Python works well but older software is not compatible.

BILLyTheLiTTle, 2018-03-30:
How can I get 1.2-SNAPSHOT from Gradle repository?

hemantmandge, 2018-04-09:
I am facing same issue with Pi4J1.1.

Unable to determine hardware version. I see: Hardware	: BCM2835
,

expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

**As suggested on forum I tried to execute program with to Pi4J 1.2-SNAPSHOT.  Error went away but GPIO output not working.** 

Can somebody suggest some other approach.

BILLyTheLiTTle, 2018-04-09:
@hemantmandge
Are you sure you are doing everything all right. Use the "gpio readall" while you are communicating with the pin and see if you have set the out mode and 1 value at the right pin. I suggest to start playing with the terminal first (http://wiringpi.com/the-gpio-utility/)

hemantmandge, 2018-04-09:
Hi,
Yes I am doing everything right.
I used program at following link which uses Pi4j. It did not work
http://agilerule.blogspot.in/2016/05/iot-spring-boot-and-pi4j-integration.html

I even tried to use sample provided with pi4j  1.2 snapshot zip. The program did not give any error as such but failed to operate GPIO pin. 

Just to give further input, I tried python program and it is working perfectly fine.

BILLyTheLiTTle, 2018-04-09:
Have you tried the wiringPi utility? Is it working with it?

qaman88, 2018-04-26:
**Thanks is now available with apt-get:**
**1- Remove existing wiringpi**
```
$ sudo apt-get purge wiringpi
$ hash -r
``` 
**2- Install wiringpi**
`$ sudo apt-get install wiringpi`
**3- Check the installation**
```
$ gpio -v
$ gpio readall
```

daabutts, 2018-08-17:
I'm having the  "I see: Hardware: BCM2835  - expecting BCM2708 or BCM2709" issue.  I've tried the same boot SD in a pi2, pi3 B, and pi3B+.  All exactly the same problem.

$ gpio readall
this seems to work and lists all the GPIO pins...

$ gpio -v
gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Embest 
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

$ pi4j -v
--------------------------------------------
            THE Pi4J PROJECT
--------------------------------------------
   VERSION   : 1.2-SNAPSHOT
   TIMESTAMP : 2018-04-23 04:22:34
--------------------------------------------

**Here are the commands I used to install:**
sudo apt-get purge wiringpi
hash -r
sudo apt-get install git-core
cd ~
git clone git://git.drogon.net/wiringPi
cd ~/wiringPi
git pull origin
./build
gpio -v
gpio readall
cd ~
pi4j -v
//this command doesn't get the latest for some reason
curl -s get.pi4j.com | sudo bash
//use this instead:
get the file from here http://get.pi4j.com/download/pi4j-1.2-SNAPSHOT.deb
sudo apt-get remove pi4j
curl -s get.pi4j.com/uninstall | sudo bash
sudo dpkg -i pi4j-1.2-SNAPSHOT.deb
pi4j -v

Here is the run command:
cmd : cd '/home/pi/NBPs//eB'; 'sudo' '/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/java' -Xdebug -Xrunjdwp:transport=dt_socket,address=127.0.0.1:53030 -Dfile.encoding=UTF-8 -Djava.library.path='lib'  -Dpi4j.debug -Dpi4j.linking=dynamic -classpath .:classes:/opt/pi4j/lib/'*'   -jar /home/pi/NBPs//eB/dist/eB.jar 
******************running Gui********************
loading
Aug 17, 2018 5:10:03 PM com.pi4j.util.NativeLibraryLoader load
FINE: Attempting to load [libpi4j.so] using path: [/lib/libpi4j.so]
Aug 17, 2018 5:10:04 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] has already been loaded; no need to load again.
Aug 17, 2018 5:10:04 PM com.pi4j.util.NativeLibraryLoader load
FINE: Library [libpi4j.so] loaded successfully using embedded resource file: [/lib/libpi4j.so]
Unable to determine hardware version. I see: Hardware: BCM2835,
 - expecting BCM2708 or BCM2709. Please report this to projects@drogon.net

The program crashes on the Java line:
final GpioController gpio = GpioFactory.getInstance();

**THEN I FOUND the problem:**  Solution:
In my NetBeans project, the Pi4J library was referencing 1.1 instead of 1.2.
Problem solved.
Hope this helps someone.

dersch81, 2018-09-28:
Had fixed this by changing the Kernel from 4.9.xx to 4.4.47. Is it fixed by newer Kernel Version?

PauloBrasko, 2018-11-21:
Solution that worked for me: I am using Netbeans, so I right-clicked in the project's name >> properties >> Categories: Run >> VM options: -Dpi4j.debug -Dpi4j.linking=dynamic -classpath .:classes:/opt/pi4j/lib/'*' 
Running again my project I got no error and the program was doing what was expected.

pvsmian, 2020-06-24:
> One other option if you need to use the existing v1.1 release of Pi4J is to use the dynamic linking option where Pi4J is dynamically linked to WiringPi rather than the default static linking.
>
> Build [...]

Would this dynamic linking work with PI4J v1.0 as well?

savageautomate, 2020-06-24:
> Would this dynamic linking work with PI4J v1.0 as well?

No, unfortunately not. 

And the custom library loader does not appear to have any logic to load alternate libs.
https://github.com/Pi4J/pi4j/blob/release/1.0/pi4j-core/src/main/java/com/pi4j/util/NativeLibraryLoader.java

pvsmian, 2020-06-24:
> No, unfortunately not.
>
> And the custom library loader does not appear to have any logic to load alternate libs. [...]

Thank you. My question btw was prompted by some migration issues going from 1.0 to 2.0, so I was trying to avoid it entirely!
- SerialDataListener changed to SerialDataEventListener, mandating code changes
- Serial.writeln(msg) isn't working the same way with a ttyxxx modem. Through trial and error, Serial.write((msg+"\r").getBytes()) seems to be the equivalent

FDelporte, 2020-06-26:
Would it make sense to add a **writeln(String msg)** method in V2 to the IODataWriter?

https://github.com/Pi4J/pi4j-v2/blob/master/pi4j-core/src/main/java/com/pi4j/io/IODataWriter.java

savageautomate, 2021-01-15:
**Closed.**  

Original issue resolved in v1.2.  

Please report any Pi4J v2 issues to the Pi4J V2 repository.
https://github.com/Pi4J/pi4j-v2/issues

closed 2021-01-15 by savageautomate