Pi4J/pi4j-v1 #320: Hardware Problem BCM2835

patriciowallace, 2017-02-24:
## With a new Raspberry pi 3 with a BCM2835, my actual system up and running reports the following:

Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

## please any solution

Azn9, 2017-02-25:
I've already started an issue with the same problem.
please close this issue and subscribe to my one.

savageautomate, 2017-02-25:
duplicate of #319

closed 2017-02-25 by savageautomate

savageautomate, 2017-08-04:
I don't know what happened to issue #319?  
So let's track against this issue now: #349