Pi4J/pi4j-v1 #325: RPIServoBlasterProvider and BCM pin numbering

salk31, 2017-03-06:
In my app I'm using BCM number elsewhere so I'd like to use it for the servo blaster too.

Of general enough interest? If so I've got a patch in mind. Also I'd like to deprecate usage of the public static members in RPIServoBlasterProvider.

natdan, 2017-03-06:
Servo blaster operates based on pin numbers. Maybe we provide both options?

salk31, 2017-03-06:
I'd quite like to add something like RaspiPhysicalPin alongside RaspiPin and RaspiBcmPin then having a mapping between them. That would then allow you to get from WiringPi or Bcm to physical to keep servod happy.

This seemed quite a bit step for me second pull request so I thought I'd just try and add RaspiBcmPin support with unit tests to make refactoring easier.

I thought even with this step I could get the unit test to sanity check the mapping by checking the wiringpi addresses match.

natdan, 2017-03-06:
I like idea of RaspiPhysicalPin sitting next to RaspiBcmPin (and RaspiPin). Are we talking about servo support or in general for pi4j?

salk31, 2017-03-06:
I though in general just with a nice side effect of making talking to servod easier.

natdan, 2017-03-06:
Yup. I am for it. Even if we don't push it in 1.2, I think it should go in!

savageautomate, 2017-03-06:
Just a note ... I don't think we maintain the actual pin mapping inside Pi4J today.  I believe we delegate that mapping responsibility to WiringPi.  I'll have to look around its been a while since I have been in that code.  WiringPi does support a physical pin mapping option.  However, we may have to add some native code to get the mapping from WiringPi when we export/provision pins.  

> **wiringPiSetupPhys (void) ;**
> Identical to above, however it allows the calling programs to use the physical pin numbers on the P1 connector only.
> 
> As above, this function needs to be called with root priviliges.
http://wiringpi.com/reference/setup/

I'm sure this change is doable, but there may be a few gotchas along the way :-)

salk31, 2017-03-07:
Ah. OK if just do BCM first? It is a step in that direction?

I was sort of assuming/hoping that one day there would be a mode that skips WiringPi to go straight to the device files. So that, as well as servod, would also require a mapping?

natdan, 2017-03-07:
My understanding is that we can allow our API to accept different type of mapping but actual implementation should stay the same (wiringPi for instance). Similarly it goes to servoblaster - internal implementation must use physical pin numbers, but this might allow it to accept any of three options. Does this make any sense at all?

salk31, 2017-03-07:
Oh yes. Strong decoupling between API and implementation. So the API hides the implementation but with a choice of numbering/naming scheme. You can then go on to swap in/out implementations. I was just suggesting that one of these implementations (one day) might be file system based...

So as my next step I was going to try and come up with a patch to make servoblaster accept WiringPi and BCM scheme to bring it up to the level of GPIO etc.. That seem reasonable?

Adding physical could be a next step. I'd write unit tests to make sure any refactor was less likely to break things.

natdan, 2017-03-07:
> _"You can then go on to swap in/out implementations. "_

What you meant by it was: _"You can then go and map supplied pin numbering to scheme that suites your implementation"_, right?

> _"So as my next step I was going to try and come up with a patch to make servoblaster accept WiringPi and BCM scheme to bring it up to the level of GPIO etc.. That seem reasonable?"_

Perfectly reasonable.

> "_Adding physical could be a next step. I'd write unit tests to make sure any refactor was less likely to break things."_

:) 👍

salk31, 2017-03-07:
>> You can then go on to swap in/out implementations.
> What you meant by it was: "You can then go and map supplied pin numbering to scheme that suites your implementation", right?

Not really. Just wanting to build on the provider class system where your code can stay the same but you just create a different provider instance. I think what you point at is a secondary advantage.

Anyway, I'll try and write some actually code for BCM ;)

natdan, 2017-03-07:
I see it now. Generally I am not against providers but to me it seems even bigger change. Would steps like this work:

1. update implementation so it uses pin mapping (according to current API and current implementation - which might be pretty much noop currently)
2. enhance servo API to allow usage of any type of pins no matter of current implementation and utilise above mechanism
3. potentially enhance other APIs to use any pin numbering scheme
4. introduce mechanism for allowing more than one implementation of the APIs (each separately of course)

I am just trying to break it down in my head so we don't pull out everything and then ended up half way to nowhere before we can start putting stuff back together. Am I closer to what you mean with these steps?

salk31, 2017-03-07:
I think you might have a grander vision than me.

For background: I'm pretty new to raspberry pi (I've only used pi 1 and pi 2) and I've never used any of the clones. I've also not fully got my head around the pi4j code. For those reasons I was trying to just follow the patterns I could see. e.g. new FooProvider(PIN_NUMBERING_SCHEME).

In your grander vision do you imagine not having to tell the provider what scheme you are using? Just pass it a Pin and it translates as/when needed?

Re "introduce mechanism for allowing more than one implementation of the APIs (each separately of course)" 
Isn't that what the providers already do?

natdan, 2017-03-07:
> _"In your grander vision do you imagine not having to tell the provider what scheme you are using? Just pass it a Pin and it translates as/when needed?"_

Yes! I though that was your idea. Maybe I skipped a few steps and went ahead! :)

> _"Isn't that what the providers already do?"_

Yes but I am not sure how far we've gone and how much more we can! I think there's quite a bit to be added across various APIs...

Anyway - if you want - give it a go how you envisaged it and we can go from there.

salk31, 2017-03-07:
I did start out on a approach closer to your grander vision but it has big problems with the API and implementation as they stand.

In the API you can ask for a list of pins that the implementation can handle. Which list would I give if the system isn't put into a particular mode?

In the implementation there is a public static Map that requires a two way mapping. Similar to the API this breaks down if there are more than one Pin instance that maps to what servoblaster is reporting.

I'm not 100% sure the grander vision is that desirable as might cause extra confusion is mixing and matching naming schemes? I think that ship might have sailed anyway given the signature of GpioProvider?

natdan, 2017-03-07:
> _"In the API you can ask for a list of pins that the implementation can handle. Which list would I give if the system isn't put into a particular mode?"_

For now we can keep 'default' - the way it is implemented. I can see easily two ways forward - and they are not mutually exclusive:

1. as you mentioned - some 'global' switch where you say what you expect and those methods to return accordingly
2. introduce parallel method with extra parameter where you pass in scheme you want returned pin to be in

Or maybe something else - you've got me now thinking and I have  loads of ideas suddenly! ;)

> _"In the implementation there is a public static Map that requires a two way mapping. Similar to the API this breaks down if there are more than one Pin instance that maps to what servoblaster is reporting."_

What about introducing some internal helper/API which will do only that - pin mapping! :)

> _"I think that ship might have sailed anyway given the signature of GpioProvider?"_

You are right, but there might be way out of it, too, without breaking a lot. Let me think about it, too. Maybe @savageautomate had some ideas over time... Pins are, after all, agnostic to the numbering scheme and numbering scheme is just way of naming particular pin (and giving its implementation way to associate it with hardware somehow)

salk31, 2017-03-07:
My other worry is/was that mapping between pins/naming might be board/revision dependent? Is it just by luck that this isn't a problem yet?

Another plus of your grand vision might be using code from different sources. In my tiny app I can choose a naming scheme and stick to it but maybe if mixing and matching that might not be so easy?

Perhaps grand vision B would be that all naming schemes would resolve so a single instance of Pin? So if you ask about Pins there would/could only be one answer?

natdan, 2017-03-07:
> _"My other worry is/was that mapping between pins/naming might be board/revision dependent?"_

Hence my idea of maybe internally make a service to do the job so that service can be then aware of hardware it works on.

> _"Perhaps grand vision B would be that all naming schemes would resolve so a single instance of Pin?"_

I can see that I jumped conclusion - I thought that was the idea you were originally talking about. :) I am sorry for confusion. Still - it is great you started the conversation - now I think more than before that it would make perfect sense for being able to map from physical/BCM/wiringPi/otherScheme to particular pins :)

salk31, 2017-03-07:
Ah. I was/am suggesting mapping between pins as a future thing. However I was a bit scared of trying to be able to use the freely without having to put Providers in particular modes.

Anyhow, I'll try and come up with a patch that, at least on balance, doesn't hinder that possible goal.

savageautomate, 2021-01-15:
**Closed.**  No further activity.

closed 2021-01-15 by savageautomate