Pi4J/pi4j-v1 #329: raspberry pi zero w Error "read(): No such device"

simondid, 2017-03-24:
hey

so i recently goth a raspberry pi zero w  until then i hade been testing my code on a pi 2 model b

on my pi 2 model b i hade no error but once i switched the sd card to raspberry pi zero w i get a bunch off terminal readouts saying

> read(): No such device

I'm gonna assume this is because the zero w is not officially supported in build 1.1 of pi4j ?

so my question is is there any way to avoid is terminal output?

and can you add some kinda pi4j identifier to the output ? as it took me quite some time to figure out that the error came from the pi4j library and not my own code  

last question is this fixed / supportet in the pi4j snapshots?

dmccue128, 2017-06-11:
I have the same problem running pi4j 1.2 - SNAPSHOT under Apache Tomcat 8.5 on Raspian OS.  
Log file fills up with endless error messages and crashes system. The repeated message is:
read(): No such device
Any idea what is causing this infinite loop?

JackieJ42, 2020-03-11:
I tracked this down to 
handleGpioPinDigitalStateChangeEvent()
Once I removed the listener from my GpioPin the messages went away and my code completed  !