Pi4J/pi4j-v1 #330: GpioPinNotProvisionedException: This GPIO pin has not been provisioned

sidf, 2017-04-03:
Hey guys. I'm getting a GpioPinNotProvisionedException exception, even though the pin seems to be provisioned. Right now I'm catching this exception, but the pin seems to be stuck in HIGH mode even though _setShutdownOptions(true, PinState.LOW)_ is called on the pin and my application is terminated.

Here's my code: https://github.com/sidf/DocumentReader/blob/master/src/main/java/io/github/sidf/documentreader/system/Lighting.java

I run an instance of Lighting in a new thread, and call the stop() method in the main thread.

Full trace:
> Exception in thread "Thread-14" com.pi4j.io.gpio.exception.GpioPinNotProvisionedException: This GPIO pin has not been provisioned: GPIO 1
>         at com.pi4j.io.gpio.impl.GpioControllerImpl.unprovisionPin(GpioControllerImpl.java:916)
>         at io.github.sidf.documentreader.system.Lighting.unprovisionPins(Lighting.java:87)
>         at io.github.sidf.documentreader.system.Lighting.stop(Lighting.java:96)
>         at io.github.sidf.documentreader.system.Lighting.start(Lighting.java:82)
>         at io.github.sidf.documentreader.system.Lighting.run(Lighting.java:67)
>         at java.lang.Thread.run(Thread.java:745)

Thanks

sidf, 2017-05-06:
It seems that unprovisionPin(Pin) fails when done from a thread that's different from the one in which provisioning happened.

Is this a bug, or a feature?

savageautomate, 2017-05-12:
Smells like a bug.  Definitely not a feature.

savageautomate, 2021-01-15:
@ganeshs4 

I realize you posed this 6 months ago, but are you still interested in getting it resolved?  I'm working on cleaning up issues for the upcoming v1.3 release.  

I tried using a modified version of your code above but was not able to reproduce the `GpioPinNotProvisionedException `.  
Does your program exit after `openOrCloseShutter()` is complete?  

I noticed a couple of potential issues ... the `GpioController ` instance probably should live outside of the function and not be shutdown until your program is ready to exit.  Also, on the last call to `pulse()` you immediately unprovision the pin.  This could be an issue since the pulse activity has not completed yet.  You could use the blocking pulse (recently added in v1.3) and wait for the pulse to complete before unprovisioning it. 

```java
garageRemoteControlOpenOrClosePin.pulseSync(2000, PinState.HIGH);
```

Here is the modified code I used to try and simulate your issue:

```java
import com.pi4j.io.gpio.*;

import java.io.IOException;

public class GpioTest {

    public static GpioController gpioGarageRemoteController = GpioFactory.getInstance();

    public static void main(String[] args) throws InterruptedException, IOException {
        boolean running = true;
        while(running){
            openOrCloseShutter();
            Thread.currentThread().sleep(100);
        }
        gpioGarageRemoteController.shutdown();
    }

    public static void openOrCloseShutter() {
        System.out.println("----------------------------------------------------------------");
        GpioPinDigitalOutput garageRemoteControlOpenOrClosePin = gpioGarageRemoteController
                .provisionDigitalOutputPin(RaspiPin.GPIO_04);

        System.out.println("Setting the pin to high i.e. open shutter on thread name: " + Thread.currentThread().getName());
        garageRemoteControlOpenOrClosePin.pulse(2000, PinState.HIGH);
        try {
            System.out.println("Thread is now going to keep looping in while loop, thread name is: "
                    + Thread.currentThread().getName());

            Thread.currentThread().sleep(3000);

            System.out.println("Thread is out of the while loop, thread name is: " + Thread.currentThread().getName());
            System.out.println("Setting the pin to pulse i.e. close shutter now");

            garageRemoteControlOpenOrClosePin.pulseSync(2000, PinState.HIGH);

            System.out.println("Shutting down the gpio controller");
            gpioGarageRemoteController.unprovisionPin(garageRemoteControlOpenOrClosePin);

        } catch (Exception e) {
            System.out.println("Error occurred while making the while loop..." + e);
        }
        System.out.println("----------------------------------------------------------------");
    }
}

```

closed 2021-02-15 by savageautomate