Pi4J/pi4j-v1 #333: ORANGE PI ZERO PIN

l-sannin, 2017-04-12:
Hello,
I test the pi4j 1.2 on my orange pi zero and only the pin 7 works, i found this wiringOp https://github.com/xpertsavenue/WiringOP-Zero
And drives all pins correctly, how can I remap or correct this on the pi4j?
Regards

savageautomate, 2017-05-12:
Pi4J's native lib for OrangePi is compiled against https://github.com/zhaolei/WiringOP 
Does https://github.com/zhaolei/WiringOP not work with OrangePi Zero? 

Here is one of the examples for using Pi4J (1.2-SNAPSHOT) with OrangePi:
https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/orangepi/GpioOutputExample.java

The current support OrangePi is still experimental and has not been fully tested.

l-sannin, 2017-05-12:
Hello, 
Thanks for your answer, I tested i2c and some GPIO pins, and some off them dont work, i found the same issue on the wiringOP C library, this because a bad pin map for the orange pi zero, but this fork https://github.com/xpertsavenue/WiringOP-Zero, drive all pins correctly, so, if you can check on this will be fantastic.
Regards

savageautomate, 2017-05-12:
Try this .... 

- Install  [WiringOP-Zero](https://github.com/xpertsavenue/WiringOP-Zero) on your RaspberryPi.
- Configure the OrangePi Platform in your code. (https://github.com/Pi4J/pi4j/blob/db2ee101b9b87d01ddf7ae30788c6a0ed04fe8cd/pi4j-example/src/main/java/orangepi/GpioOutputExample.java#L63)
- Configure pi4j to use dynamic linking `java -Dpi4j.linking=dynamic`.

When you start your Java app using the system property "pi4j.linking=dynamic", Pi4J will bypass its embedded native WiringPi bindings and attempt to dynamically link to your WiringXXX .SO libraries installed on your system.  

Thanks, Robert

l-sannin, 2017-05-13:
Hello,
I have test, and now works like a charm!!!.
Thanks a lot Robert

jkandasa, 2017-08-29:
To avoid confusion on @savageautomate workaround,

Install [WiringOP-Zero](https://github.com/xpertsavenue/WiringOP-Zero) on your ~~RaspberryPi~~ **OrangePi Zero**.

savageautomate, 2017-09-05:
@jkandasa 
Thanks for pointing out the mistake, I have updated the original entry to avoid confusion.

savageautomate, 2021-01-15:
**Closed.**  No further activity

closed 2021-01-15 by savageautomate