Pi4J/pi4j-v1 #334: Invalid pin exception using CM3 with i/o board

ninja-asa, 2017-04-19:
Hello,

I had been using to Raspberry 3 B to control some GPIO and taking some picture using the camera module and just recently moved on to using a CM3 with an i/o board. For the GPIO control, I have been using pi4j and, with the Compute Module now it gives the following error:
Exception in thread "main" com.pi4j.io.gpio.exception.InvalidPinException: Invalid pin exception; this pin [GPIO 32] is not supported by GPIO provider [RaspberryPi GPIO Provider]
Using any of the pin from the J6 header.

I tried using the pin from the J5 header but apparently the it is not activating/reading the correct pins... when I code in c++ and use wiringPi the behaviour I expect occurs, When

  private final GpioController gpio = GpioFactory.getInstance();
  ONE = gpio.provisionDigitalOutputPin(RCMPin.GPIO_32, "Output1", PinState.LOW);
  TWO = gpio.provisionDigitalOutputPin(RCMPin.GPIO_05, "Output2", PinState.LOW);
  THREE = gpio.provisionDigitalOutputPin(RCMPin.GPIO_06, "Output3", PinState.LOW);
  FOUR  = gpio.provisionDigitalInputPin(RCMPin.GPIO_07, "Input1");

Any idea?

wborn, 2019-02-07:
I ran into the same issue with 1.2-SNAPSHOT (required to fix https://github.com/Pi4J/pi4j/issues/319) and a CM3 (running latest Raspbian). 

The `InvalidPinException` could be resolved by using Broadcom GPIO numbers instead:

```java	
GpioFactory.setDefaultProvider(new RaspiGpioProvider(RaspiPinNumberingScheme.BROADCOM_PIN_NUMBERING));
```

savageautomate, 2021-01-15:
**Closed.**

This issue has been fixed in latest v1.3-SNAPSHOT build.  The GPIO pins were not getting detected and exported properly for CM3 and CM3+.

I tested and validated input listeners for all GPIO pins (0-45) on a CM3 running the latest RaspberryPi OS, WiringPi v2.60 and Pi4J v1.3-SNAPSHOT.

commit ba5d7050 (2021-01-15): Fix #334; Fix #406; Invalid pin exception using CM3. CM3 and CM3+ pins were not getting detected and exported properly.
commit 4eb14fc1 (2021-01-15): Fix #334; Fix #406; Invalid pin exception using CM3. (previous commit 'ba5d705' failed to compile on non-RPI boards)
closed 2021-01-15 by savageautomate