Pi4J/pi4j-v1 #336: How to use P4J without locking into any platform?

malkusch, 2017-04-24:
Hi

When I follow [this example](http://pi4j.com/example/listener.html), I will need an instance of `Pin`. The example proposes `RaspiPin.GPIO_02`. Now I'm wondering how I can abstract the platform away. `PinProvider.getPinByAddress(2)` doesn't do the job corrrectly. If I'm understanding the code correctly, the underlaying map is hard coupled to any of those platform specific PinProvider instances. And the map is even shared. So even a simple `switch(platform)` would not create predictable results, as it totally depends on how the classloader loaded the PinProvider classes and how HashMap orders them.

My current dirty workaround is subclassing PinProvider which gives me access to createDigitalPin(), where I can use `PlatformManager.getPlatform().getGpioProvider().getName()`. But this really looks, like someone accidentially put that protected while it should be package protected and also I'm loosing all the platform specifics from each PinProvider.

So how would I get a `Pin` instance for address 2 without locking to a platform? Is there something I'm missing, or is this use case not supported? If latter, let's change that.

savageautomate, 2017-05-12:
Hi @malkusch 

I see what you are saying.  The relationship between the Pin and PinProvider do make it difficult to abstract.

I like what you are suggesting about adding "`Platform.getPinProvider()`" however, this is not explicitly a one-to-one relationship.  In the case of the RaspberryPi platforms, there are currently three Pin/PinProvider instances on that platform.  We could do something like this:

```
- Platform.getPinProviders()
- Platform.getDefaultPinProvider()
```

We would have to make assumptions about a default pin provider.

The closest example to abstracting the pin acquisition logic I found is in this method:
https://github.com/Pi4J/pi4j/blob/db2ee101b9b87d01ddf7ae30788c6a0ed04fe8cd/pi4j-core/src/main/java/com/pi4j/util/CommandArgumentParser.java#L47

However, it still requires knowledge about the PinProvider class -- so not really abstracted at all, just reflectively getting the pin instance from the provider class instead of a concrete PinProvider instance. 

Thoughts?

thegreystone, 2017-11-01:
Why not always initialize the wiring pi library in RaspiPin-mode, and have different implementations of the pins (Raspi/Broadcom etc) mapping to different RaspiPin addresses? Then you could use all kinds of pins interchangeably and simultaneously.

savageautomate, 2021-01-14:
Closed; Pi4J v2.0 provides a new pluggable model for platform abstraction.

closed 2021-01-14 by savageautomate