Pi4J/pi4j-v1 #337: Support Sinovoip's Banana Pi M1+

malkusch, 2017-04-24:
The statically linked Banana Pro wiringpi library doesn't work with Banana Pi M1+. This is no surprise as the BPI-M1+ uses a different [pin numbering](http://linux-sunxi.org/GPIO#Accessing_the_GPIO_pins_through_sysfs_with_mainline_kernel). However it took me some time to understand that and also to find a solution. So I like to abuse this issue mainly for documentation:

Until pi4j supports a further platform which is linked against [BPI-WiringPi](https://github.com/BPI-SINOVOIP/BPI-WiringPi) you can still use Pi4j with the Banana Pi M1+.
- Install  [BPI-WiringPi](https://github.com/BPI-SINOVOIP/BPI-WiringPi).
- Configure the [Banana Pro platform](http://pi4j.com/pins/lemaker-bananapro.html#Platform_Configuration)
- Configure pi4j to use dynamic linking `java -Dpi4j.linking=dynamic`.

savageautomate, 2017-08-07:
@malkusch,

**UPDATE**
I have just added experimental support for the Sinovoip BPI series "BananaPi" variants.

COMMIT: https://github.com/Pi4J/pi4j/commit/92f06587e92d740d78a98ff45881c6ae44604316

This is included in the the latest 1.2-SNAPSHOT build now.

Please note that I do not have any of these devices so I'm not able to really test it.
I have included a new "bpi" platform that you have to choose when starting your program.

```
        // ####################################################################
        //
        // since we are not using the default Raspberry Pi platform, we should
        // explicitly assign the platform as the Sinovoip BPI platform.
        //
        // ####################################################################
        PlatformManager.setPlatform(Platform.BPI);
```

**STATUS:**

- I have added support for building the BPI-WiringPi library (https://github.com/BPI-SINOVOIP/WiringPi) and embedding them inside the Pi4J JARs.  
- I have created a new BpiGpioProvider (https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/BpiGpioProvider.java) and BpiPin (https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/BpiPin.java) class that should be used.
- The BpiPin class is purely a copy from the BananaPro class, I have not verified the pin mappings with the BPI devices. 
- I have not tested anything (other than the build)
- I have not added any samples to the examples yet.
- Board detection (M1, M2, M3, M1+, M2+, M2P, etc) has not been added yet. 

Feel free to start testing it and report back any issue .. or better yet submit pull requests with fixes.

Thanks, Robert

savageautomate, 2021-01-15:
**Closed.**  

As of Pi4J v2.x, board/platform expansion will provided via pluggable modules and support for specific platforms beyond the RPI is outside of the scope of the Pi4J project.   We had to make this change because supporting and testing 'N' number of board variants became impossible for a small volunteer team to keep up with.  Moving forward, Pi4J will provide the core framework and infrastructure and if someone wants to create _drivers_ for alternate platforms/boards, they will be able to do that and maintain it separately from the Pi4J codebase.  

Pi4J v2 also does not mandate/assume an implementation like WiringPi.

See: https://v2.pi4j.com/documentation/architecture

closed 2021-01-15 by savageautomate