Pi4J/pi4j-v1 #339: 3wire SPI (bidirectional mode) support for RPi 3

DoctorZIP, 2017-05-14:
As mentioned [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md) RPi3 has 3wire SPI support. Is it possible to implement in in PI4J?

savageautomate, 2017-08-04:
At the moment the SPI implementation in Pi4J is based on WiringPi's SPI API.  

This thread suggests that the 3-wire needs some special configuration set for the spidev device.
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=132426
```
ioctl(fd, SPI_IOC_RD_MODE32, 0x10); //enable 3wire mode
```
I looked at the WiringPi SPI implementation code and did not readily see any support for configuring this with the existing API.   
https://github.com/Pi4J/wiringPi/blob/master/wiringPi/wiringPiSPI.c

So we have to wait until WiringPi supports it or port Pi4J's SPI impl to use spidev directly in the native library.  So yes .. its possible, just not simple or quick to get done :-) 

Thanks, Robert

savageautomate, 2021-02-15:
No further activity; issue closed.

closed 2021-02-15 by savageautomate