Pi4J/pi4j-v1 #341: tomcat startup faild  wiringPiSetup: Must be root. (Did you forget sudo?)

jwister, 2017-05-25:
tomcat startup faild  wiringPiSetup: Must be root. (Did you forget sudo?) 
 why?   the curent User  is root

natdan, 2017-05-25:
How did you start tomcat?

jwister, 2017-05-25:
use  "service tomcat7 start"

jwister, 2017-05-25:
My current user is "root"

natdan, 2017-05-25:
Ah - but current user does not affect which user tomcat service works as ... If you check /etc/init.d/tomcat file (or whatever is there) you might see something like
/bin/su $TOMCAT_USER -c $CATALINA_HOME/bin/startup.sh
where $TOMCAT_USER is set to something else... Possibly 'tomcat'? If you want to run tomcat as a 'root' you'll probably need to change it in there. 

So, check /etc/init.d/tomcat and try to change $TOMCAT_USER in there to 'root'. Let me know if that fixed your problem.

jwister, 2017-05-25:
Changed the "root", but any of the same mistakes, are I wrong?
[tomcat7.txt](https://github.com/Pi4J/pi4j/files/1029527/tomcat7.txt)

natdan, 2017-05-25:
Huh. No idea. I would say that's exactly what you wanted. But, again - there you might need to do something before changes are picked up. For instance try: 
> sudo systemctl daemon-reload
> sudo service tomcat stop
> sudo service tomcat start

And last - check if tomcat is really run as root by:
> ps aux | grep tomcat

jwister, 2017-05-25:
A little headache    ， See the picture
[image]

natdan, 2017-05-25:
Run what it is telling you: 
> systemctl status tomcat7.service
Post result here if you're unsure what's the issue after reading it! :)

jwister, 2017-05-26:
The problem is solved！  The configuration file "etc / default / tomcat7" also modifies users and groups。

thank you for your help。

natdan, 2017-05-26:
👍

closed 2017-05-26 by natdan