Pi4J/pi4j-v1 #345: Unkown board with Pi 0 W

tuytuy, 2017-06-14:
**Error**
` ==>> SERIAL SETUP FAILED : Unsupported/unknown board type; unable to detect hardware.`

I have tried the exact same code on a Pi 3 and it all worked fine. To do so I just removed the sd card from the Pi 0W and put it in the Pi 3 and it all worked as expected.

What could be done about this ? I am using the latest realsed version of the library, not the SNAPSHOT.

Thanks

savageautomate, 2017-06-14:
Try the SNAPSHOT build.  It has a newer WiringPi revision compiled in.

savageautomate, 2017-08-04:
Closing this issue.

I tested the SerialExample.java program using Pi4J version 1.2-SNAPSHOT today on RPi Zero W and it seemed to work fine.  

Re-open if you continue to experience any problems.

Thank You, Robert

closed 2017-08-04 by savageautomate