Pi4J/pi4j-v1 #346: Problems Pi4J, RPi BCM2835

Edge-Walker, 2017-06-20:
The following code compiles, but fails at runtime when trying to create the GPIO controller instance with the error. The hardware number is correct. WiringPi documentation claims that version 2.44 supports this hardware.

> Unable to determine hardware version. I see: Hardware	: BCM2835
> ,
>  - expecting BCM2708 or BCM2709.
> If this is a genuine Raspberry Pi then please report this
> to projects@drogon.net. If this is not a Raspberry Pi then you
> are on your own as wiringPi is designed to support the
> Raspberry Pi ONLY.

```
// Import Pi4J libraries
import com.pi4j.io.gpio.GpioController;
import com.pi4j.io.gpio.GpioFactory;
import com.pi4j.io.gpio.GpioPinDigitalOutput;
import com.pi4j.io.gpio.PinState;
import com.pi4j.io.gpio.RaspiPin;

public class StepperControl {
    public static void main(String args[]) {`
        System.out.println("Hello world!");

	// Create a GPIO controller instance
	final GpioController gpio = GpioFactory.getInstance()
}
```

Edge-Walker, 2017-06-20:
This is a duplicate of [(https://github.com/Pi4J/pi4j/issues/319)0

closed 2017-06-20 by Edge-Walker

chazzG, 2017-07-05:
319 is gone, can't see the page

dionusos, 2017-07-29:
319 is still unavailable.

savageautomate, 2017-08-04:
I don't know what happened to issue #319?  
So let's track against this issue now: #349