Pi4J/pi4j-v1 #347: SystemInfo.getClockFrequencyCore() Not woking in 1.1SNAPSHOT Orange Pi Zero

Subhajitdas298, 2017-06-22:
Exception generated is :
java.io.IOException: Cannot run program "/opt/vc/bin/vcgencmd": error=2, No such file or directory

savageautomate, 2017-08-04:
Which operating system/distribution are you using?  

I think I tested on Raspbian for OrangePi when I first ported this over.
http://www.orangepi.org/downloaded/download.html

Thanks, Robert

Subhajitdas298, 2017-08-04:
I tested it on armbian.
Is there any way to add "vcgencmd"?
Or at least implement all of it's features using other common linux command.
"vcgencmd" seems to be raspberry pi specific.

savageautomate, 2017-08-04:
I don't know about adding "vcgencmd" -- I think you are right about that being specific to Raspberry Pi.

Are you specifying the platform with Pi4J for OrangePi?
Example:
https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/orangepi/GpioInputExample.java#L71

```
PlatformManager.setPlatform(Platform.ORANGEPI);
```

After a quick look at the code, I don't think you should be getting that error.  For OrangePi you should be getting an "**UnsupportedOperationException**" exception thrown.  This indicates that the platform implementation does not support that function.

Subhajitdas298, 2017-08-05:
Well I will set platform next time. But "UnsupportedOperationException" isn't a solution.
At least implement all of it's features using other common linux command, if possible. Or guide me, I would like to help test this for you. Know any sources?

savageautomate, 2021-02-15:
Orange Pi Zero is no longer supported in Pi4J v1.4. 
For more details, please see:
 - https://pi4j.com/1.4/index.html
 - https://github.com/Pi4J/pi4j/blob/master/README.md

closed 2021-02-15 by savageautomate