Pi4J/pi4j-v1 #348: Error writing to I2CDevice on I2CBus '1' ('/dev/i2c-1')

nitipongs, 2017-06-27:
I  am using  raspberry pi 3  with raspain lite  kernel 4.1 installed  and  I2C enabled 
pi4j 1.1 installed 
i2cdetect -y 1  found  I2C  device  at  0x57

my code

public static final int I2C_ADDR = 0x57; // address pin not connected (FLOATING)

    public static void main(String[] args) throws InterruptedException, PlatformAlreadyAssignedException, IOException, UnsupportedBusNumberException {

        I2CBus i2c = I2CFactory.getInstance(I2CBus.BUS_1);
        I2CDevice device = i2c.getDevice(I2C_ADDR);
        Thread.sleep(500);
        byte[] cmd_values = new byte[]{0x01,0x0A, 0x00, 0x01,0x02 , 0x12 , 0x01 ,0x00, 0x02 ,(byte)0xDC};
        device.write(cmd_values);
    }

and here  is  an error

Exception in thread "main" java.io.IOException: Error writing to I2CDevice on I2CBus '1' ('/dev/i2c-1') at address 0x57. Got '-20001'.
	at com.pi4j.io.i2c.impl.I2CDeviceImpl.write(I2CDeviceImpl.java:111)
	at com.pi4j.io.i2c.impl.I2CDeviceImpl.write(I2CDeviceImpl.java:124)
	at I2COsw.main(I2COsw.java:40)

nitipongs, 2017-06-28:
the problem  has been fixed  by  increased  sleep  from  500  to  3000

natdan, 2017-06-29:
3000 is incredibly high. Maybe it is worth making tiny example in python and see if the same would happen there. So far I cannot see why would Java impl have an issue with discovering i2c slave. 

Other thing is - what you mentioned - maybe deliberately pull floating pin to ground. Theoretically it is possible that it really floats and it takes time to settle down when you first access the chip.

BTW no matter how badly the error conveys the message - message is 'there's no device at that address' O: )

nitipongs, 2017-07-01:
Which  floating pin  you  recommend  to  pull  down  ,   I2C  device  or  Raspi ?

I  still  experience  stability  issue  after  run  it  for  a  while , maybe  an hour ,  it encountered  io error  and  stop  working again.