Pi4J/pi4j-v1 #349: Pi4J fails with Rpi 4.9.35  kernel - Unable to determine hardware version from wiringpi lib

chazzG, 2017-07-05:
It seems the Rpi 4.9 kernel always puts BCM2835 into /proc/cpuinfo. The wiringpi static library linked with the latest release of Pi4J fails. I checked the source code of wiringPi.c and the 2.44 version should work with Pi4J, as the code allows BCM2708, BCM2709, and BCM2835 as hardware types. One can not simply recompile wiringPi as the shared libs are not used by Pi4J, they are static linked into Pi4J. I hope this gives u enough info to build a new release. I began to recompile Pi4j with the new wiringPi 2.44, but maven fails and also loaded more usuless crap into my little old pi than I could tolerate. Thanks for your support!

ameenmohamed, 2017-07-09:
having same issue

ameenmohamed, 2017-07-09:
I posted question to the email that came up in error message i got this reply 
Standard reply as I'm fed-up typing it in every time.

So, you've upgraded to a shiny new 4.9 kernel and wiringPi has stopped
working.

The simple answer is: Upgrade wiringPi.

Use

  sudo apt-get update
  sudo apt-get upgrade

mechanism.

However...

If you are running Python/PHP/Java, or some language other than C/C++
or BASIC, or some pre-built system such as a TV/Video release or home
automation release,  then sorry - you're screwed.

wiringPi is a C library for C/C++ and BASIC programs. I do not support
anything else.

So the morons who put together the wrappers for the language you're using,
or the system you have have been complete utter twats and statically
linked an old version of wiringPi into their system. I fixed this problem
in December 2016 in anticipation of the 2.9 kernel being released, but
it seems these incompetent buffoons haven't bothered to upgate their
own system that you now have the misfortune to use.

Get in-touch with them, preferably with a very sharp stick and get them
to fix the problem.

This may help for Java issues:

  https://github.com/Pi4J/pi4j/issues/319

and this may help Python issues:

  https://github.com/WiringPi/WiringPi-Python/issues/47

Good luck.

chazzG, 2017-07-09:
> WiringPi/WiringPi-Python#47

I got it working by

downloading the pi4j 1.2 SNAPSHOT, installed it, and problem is resolved
with that release. the download is on the Pi4j website

that download is linked to the correct version of wiringPi library

Chuck

Good Luck

savageautomate, 2017-07-09:
**The Pi4J SNAPSHOT version is (_almost_) always compiled against the latest WiringPi release.**

We choose early on to statically link against WiringPi (by default) as a number of Java programmers struggled with maintaining WiringPi releases and managing that at a dependency for distribution.  It also ensured that we could ship a release of Pi4J tested against a known stable release of WiringPi and reducing a lot of frustration when the two libs would get out of sync with ether other on a users's system.  

All that being said, its unfortunate that this error message refers uses to seek support from WiringPi (Gordon) when his library was fixed ages ago.  Perhaps we can implement some error handling to provide a better/more helpful error message on our end.  

Pi4J also optionally supports dynamic linking against WiringPi for uses or use cases when a newer or custom version of WiringPi is needed.  All you have to do is set the system property:
`pi4j.linking=dynamic.`

I have been very busy in other parts of my life and have not been as active on this project as I would like and I really need to get a release of 1.2 tested and published.  Hopefully very soon.

Thanks, Robert

GeVanCo, 2017-07-15:
I hit the same stone a couple of days ago.  After some investigation, I found the reason as to why wiringPi didn't work anymore.

The reason behind this is that with kernel 4.8+ the name of the processor has changed from BCM2708/BCM2709 into BCM2835.  If you have the RPi up and running, check the /proc/cpuinfo content and you'll see BCM2835 next to the "Hardware" line.  Revision remained the same.

The previous versions of wiringPi were only checking for the presence of BCM2708 or BCM2709 into that 
file and therefore, with the new kernel installed, that check failed.  Hence, exit_failure was given.
wiringPi has been updated to tackle that issue: it now checks for BCM2708/09 (to remain backwards compatible with older kernels) as well as for BCM2835.  That solves the issue.

If you want to see the details, pls. check for wiringPi.c, line 715 and further in the latest wiringPi repo (which is downloaded into Pi4J if you also take the native build for RPi along).  It now looks like this:

`
// See if it's BCM2708 or BCM2709 or the new BCM2835.

// OK. As of Kernel 4.8,  we have BCM2835 only, regardless of model.
//	However I still want to check because it will trap the cheapskates and rip-
//	off merchants who want to use wiringPi on non-Raspberry Pi platforms - which
//	I do not support so don't email me your bleating whinges about anything
//	other than a genuine Raspberry Pi.

  if (! (strstr (line, "BCM2708") || strstr (line, "BCM2709") || strstr (line, "BCM2835")))
  {
    fprintf (stderr, "Unable to determine hardware version. I see: %s,\n", line) ;
    fprintf (stderr, " - expecting BCM2708, BCM2709 or BCM2835.\n") ;
    fprintf (stderr, "If this is a genuine Raspberry Pi then please report this\n") ;
    fprintf (stderr, "to projects@drogon.net. If this is not a Raspberry Pi then you\n") ;
    fprintf (stderr, "are on your own as wiringPi is designed to support the\n") ;
    fprintf (stderr, "Raspberry Pi ONLY.\n") ;
    exit (EXIT_FAILURE) ;
  }
`

savageautomate, 2017-08-04:
_This is an update that I posted to a RaspberryPi forum thread and am including it here for reference._
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=182191

**For clarification .....**

This error is an error message output by the WiringPi library due to a change in the hardware signature starting with kernel version 4.8.   

You can run the following command and see the "Hardware" information displayed by the kernel.

```
cat /proc/cpuinfo
...
Hardware	: BCM2835
```
However, Gordon updated WiringPi to support this change long ago (_December 2016_) in WiringPi version 2.36.  
(commit: http://git.drogon.net/?p=wiringPi;a=commitdiff;h=b1dfc186efe327aa1d59de43ef631a2fa24e7c95)

And Pi4J included these updates from WiringPi somewhere around January 2017 in Pi4J version 1.2-SNAPSHOT builds.

**So why does this issue continue to popup?   And why does installing the latest WiringPi on my system not resolve the issues?**

Early on in Pi4J development we started embedding (_static linking_) a pre-compiled version of WiringPi inside the native Pi4J package.  We did this because at that time WiringPi was not included in the OS image and many Java programmers new to the embedded world struggled with the WiringPi dependency and making sure compatible versions of Pi4J and WiringPi were in sync.  

So (_by default_) Pi4J does not care or use what version of WiringPi you may have installed on your system.  Thus if you are using Pi4J version 1.1 or lower, then you are still using an outdated version of WiringPi.  **The solution is to upgrade to Pi4J version 1.2-SNAPSHOT**.   Most users have reported success after upgrading to Pi4J version 1.2-SNAPSHOT.  However a small number of users have reported still having this issue after upgrading.  For users continuing to experience this issue, I'm not sure what's going wrong and would be happy to help if you can provide some further details on your environment and steps you have attempted to solve the problem.  

Another workaround users can try is to tell Pi4J not to use it's embedded WiringPi and instead use the shared WiringPi library installed on your system.  Pi4J does optionally support dynamic linking against WiringPi for use cases when a newer or custom version of WiringPi is needed. All you have to do is set the system property:

`> pi4j.linking=dynamic`

You can also include this in your java command line when launching your java program with the "`-D`" argument:

`java -Dpi4j.linking=dynamic`

Perhaps the biggest help would be to get Pi4J version 1.2 released.  Its way overdue.  I have been pulled in many directions since the start of 2017 and have not been able to focus on the Pi4J project as much as I would like to.  I will make an effort to get version 1.2 released soon.     

Thank You,
Robert

robertsbd, 2017-08-12:
I received the same issue as all of you using the Pi4J lib in a clojure project on the rasp pi. I switched my project dependencies to use "https://oss.sonatype.org/content/groups/public" and the library being 1.2-SNAPSHOT and it is no longer reporting the error. So I was able to get away without having to download the libraries.

Thanks for the help everyone.

Bill5203, 2017-09-17:
uname -srvo
Linux 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 GNU/Linux

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

Pi4J Snapshot was installed as it is instructed as instructed by Robert:

sudo apt-get remove pi4j
sudo dpkg -i pi4j-1.2-SNAPSHOT.deb

Still doesn't work so taking next advice:
"Pi4J does optionally support dynamic linking against WiringPi for use cases when a newer or custom version of WiringPi is needed. All you have to do is set the system property":

pi4j.linking=dynamic
-bash: pi4j.linking=dynamic: command not found

Now what?

Bill

eitch, 2017-09-18:
Hey Bill. pi4j.linking should probably be set as a system property. So try this:

    $bash: export pi4j.linking=dynamic

thegreystone, 2017-10-25:
Any ETA on 1.2?

thegreystone, 2017-10-25:
If not, is there anything blocking the release that the community can help out with?

mirage22, 2017-10-31:
are there any plans about to have version 1.2 released ?

jCecchetti, 2017-11-18:
I had this same problem. Everything was solved after switching to 1.2-SNAPSHOT

DRChess, 2017-12-07:
What happens if the OS calls out the correct version of the CPU in a Pi3 as BCM2837 (1.1) or version BCM2838 (1.2) instead of the generics for Broadcom's A53/Video core IV ??? Or maybe the OS can't see that deep into the chip design. It would have been better to lockout the Pi knock off with just looking for the string with BCM.

acrossi, 2017-12-13:
Hoping for a 1.2 for Christmas!

I have been so good this year....

Kstarks87, 2017-12-14:
Can someone provide the exact commands to install the version 1.2 SNAPSHOT? I'm a noob and have run into this unable to determine hardware version issue.

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

ChrisCarrAu, 2018-05-06:
I don't know if this will help, but here are my workings to get it all going
https://github.com/ChrisCarrAu/RaspberryPi-IoT/wiki

savageautomate, 2018-05-06:
@BILLyTheLiTTle 

1.2-SNAPSHOT builds are available in this repository:
http://pi4j.com/download.html#Maven_Repository

savageautomate, 2018-05-06:
This issue should be fully resolved in the latest 1.2-SNAPSHOT builds. 
Note, you must have the latest WiringPi installed on your Raspberry Pi separately from Pi4J 1.2-SNAPSHOT.

closed 2018-05-06 by savageautomate