Pi4J/pi4j-v1 #350: Armbian Mainline Banana pi M1 eth0 problem

noudje, 2017-07-27:
If I try to use the latest mainline armbian in combination with PI4J and use one of the pins also used for SPI as GPIO then the eth0 gets offline.

The problem GPIO pins are: 10, 11, 12, 13 and 14.
kernel: Linux bananapi 4.11.6-sunxi #6 SMP Fri Jun 23 19:56:18 CEST 2017 armv7l GNU/Linux 
platform: Banana PI M1

I've also posted example code of the problem at the armbian forum. 
https://forum.armbian.com/index.php?/topic/4804-mainline-debian-armbian-banana-pi-pi4j/&_fromLogin=1

Is there a way to quick fix my problem?
regards,
Arnoud

savageautomate, 2017-08-04:
Hi @noudje,

I don't think I fully understand.  So .. if you attempt to use a GPIO pin that is shared with SPI then it causes the network connection to do down.  

I don't have an M1 to test with, only a LeMaker BananaPi.    
Is LeMaker WiringBP compatible with BPI M1?   Pi4J is using WiringBP for all the low level stuff.

https://github.com/LeMaker/WiringBP

I suspect maybe some incompatibility at this layer without knowing more details.

You might try using the WiringBP directly and see if the problem is still there.  If WiringBP works perfectly, then we would know that its a problem somewhere higher in the Pi4J layer.

Thanks, Robert

savageautomate, 2017-08-04:
Also take a look at issue #337.  @malkusch describes a problem with the pin numbering differences between BananaPro and M1+.   Maybe his suggestion about using BPI-WiringPi and dynamically linking it to Pi4J would work for your case too?

savageautomate, 2017-09-05:
Experimental support has been added for BPI versions of BananaPi.
See https://github.com/Pi4J/pi4j/issues/337

Closing this issue for now, please re-open if this problem still exists in the new BPI supported version.

closed 2017-09-05 by savageautomate