Pi4J/pi4j-v1 #352: RPI 2 seems to "Hangs" or "Crash" after setting pinmode to PWM output

ribasco, 2017-08-01:
Been stuck on this problem the whole day.

I'm simply trying to test PWM output to my piezo buzzer. I did some tracing and found that the Pi  seems to "Hang/Crash" after a call to `com.pi4j.wiringpi.Gpio.pinMode();` (WiringPiGpioProviderBase.java, Line 155)

I tried with both of these methods (see below) and I get the same result. Its hard to say whether it crashed or hangs, but my SSH connection gets dropped after 30 seconds or so and the Pi no longer seems to respond.

`GpioPinPwmOutput buzzer = gpio.provisionPwmOutputPin(RaspiPin.GPIO_26, "Buzzer");`
	or 
`Gpio.pinMode(RaspiPin.GPIO_26.getAddress(), PWM_OUTPUT);`

However, If I set the pinmode from the command line `gpio mode 26 pwm` and do a '`gpio pwm 26 500`' for example, it works.	

Further info on my environment setup:

RPI 2 Model B
Pi4j 1.2-SNAPSHOT
Kernel v4.9.35-v7+
Wiring pi v2.44

ribasco, 2017-08-01:
Just read a post in https://github.com/WiringPi/WiringPi-Python/issues/47. 

Tested it with 'sudo' now everything is working....Derp..

closed 2017-08-01 by ribasco

MrCustomizer, 2017-12-28:
I think this issue can be reopened. There has been some movement concerning root privileges and PWM functionality in Raspbian. Seems like the following patch makes it possible to use hardware PWM without sudo:
https://github.com/raspberrypi/linux/issues/1983

Perhaps it could be included in the non-privileged mode in an upcoming release? I really would like to see that.