Pi4J/pi4j-v1 #353: java.io.IOException: I2CBus '1' ('/dev/i2c-1') has already been closed!

ribasco, 2017-08-06:
I have pins on the MCP23017 where I set it's shutdown options. So it seems that the error is thrown when pi4j tried to reset the pin state on shutdown event. 

**Error:** 
```
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.RuntimeException: java.io.IOException: I2CBus '1' ('/dev/i2c-1') has already been closed! A new bus has to be acquired.
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1895)
	at com.ibasco.rpi.usbps.app.I2CTest.main(I2CTest.java:468)
Caused by: java.lang.RuntimeException: java.io.IOException: I2CBus '1' ('/dev/i2c-1') has already been closed! A new bus has to be acquired.
	at com.pi4j.gpio.extension.mcp.MCP23017GpioProvider.setState(MCP23017GpioProvider.java:266)
	at com.pi4j.io.gpio.impl.GpioPinImpl.setState(GpioPinImpl.java:325)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.shutdown(GpioControllerImpl.java:1036)
	at com.ibasco.rpi.usbps.app.I2CTest.shutdown(I2CTest.java:488)
	at java.util.concurrent.CompletableFuture.uniWhenComplete(CompletableFuture.java:760)
	at java.util.concurrent.CompletableFuture$UniWhenComplete.tryFire(CompletableFuture.java:736)
	at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474)
	at java.util.concurrent.CompletableFuture$AsyncRun.run(CompletableFuture.java:1632)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: I2CBus '1' ('/dev/i2c-1') has already been closed! A new bus has to be acquired.
	at com.pi4j.io.i2c.impl.I2CBusImpl.testForProperOperationConditions(I2CBusImpl.java:297)
	at com.pi4j.io.i2c.impl.I2CBusImpl.runBusLockedDeviceAction(I2CBusImpl.java:251)
	at com.pi4j.io.i2c.impl.I2CBusImpl.writeByte(I2CBusImpl.java:185)
	at com.pi4j.io.i2c.impl.I2CDeviceImpl.write(I2CDeviceImpl.java:131)
	at com.pi4j.gpio.extension.mcp.MCP23017GpioProvider.setStateA(MCP23017GpioProvider.java:282)
	at com.pi4j.gpio.extension.mcp.MCP23017GpioProvider.setState(MCP23017GpioProvider.java:261)
	... 14 more
```

**Test Code:**

```
//Called from the constructor
private void setupPins() {
        //Relays
        relay1 = gpio.provisionDigitalOutputPin(mcpProvider, MCP23017Pin.GPIO_A2, "RELAY - 1");
        relay2 = gpio.provisionDigitalOutputPin(mcpProvider, MCP23017Pin.GPIO_A3, "RELAY - 2");
        relay1.setShutdownOptions(true, PinState.LOW);
        relay2.setShutdownOptions(true, PinState.LOW);

        //LCD
        lcd_backlight = gpio.provisionDigitalOutputPin(mcpProvider, MCP23017Pin.GPIO_B6, "LCD - Backlight");
        lcd_backlight.setShutdownOptions(true, PinState.LOW);

        //ADC - Sensors
        currentAnalogInput = gpio.provisionAnalogInputPin(adcProvider, ADS1115Pin.INPUT_A3, "Current Sensor");
        //Buttons
        GpioPinDigitalInput gpioInButton1 = gpio.provisionDigitalInputPin(RaspiPin.GPIO_06, "Back", PinPullResistance.PULL_UP);
        GpioPinDigitalInput gpioInButton2 = gpio.provisionDigitalInputPin(RaspiPin.GPIO_05, "Previous", PinPullResistance.PULL_UP);
        GpioPinDigitalInput gpioInButton3 = gpio.provisionDigitalInputPin(RaspiPin.GPIO_04, "Next", PinPullResistance.PULL_UP);
        GpioPinDigitalInput gpioInButton4 = gpio.provisionDigitalInputPin(RaspiPin.GPIO_01, "Select", PinPullResistance.PULL_UP);

        //Initialize Buttons
        button1 = new GpioButtonComponent(gpioInButton1, PinState.HIGH, PinState.LOW);
        button1.setName("Back");

        button2 = new GpioButtonComponent(gpioInButton2, PinState.HIGH, PinState.LOW);
        button2.setName("Previous");

        button3 = new GpioButtonComponent(gpioInButton3, PinState.HIGH, PinState.LOW);
        button3.setName("Next");

        button4 = new GpioButtonComponent(gpioInButton4, PinState.HIGH, PinState.LOW);
        button4.setName("Select");
    }
```

```
public static void main(String[] args) throws Exception {

        I2CTest test = new I2CTest();
        log.info("Starting Test...");
        ArrayList<CompletableFuture> l = new ArrayList<>();
        l.add(CompletableFuture.runAsync(test::runI2CLcdTest, es));
        log.info("Started LCD Test");
        l.add(CompletableFuture.runAsync(test::runButtonInputTest, es));
        log.info("Started Button Test");
        CompletableFuture.allOf(l.toArray(new CompletableFuture[l.size()])).whenComplete(test::shutdown).get();

        log.info("Shutdown complete!");
    }

    private void shutdown(Void p, Throwable t) {
        if (t != null) {
            log.error(t.getMessage(), t);
            return;
        }
        adcProvider.shutdown();
        mtService.shutdown();
        es.shutdown();
        gpio.shutdown();
        //mcpProvider.shutdown();
    }
```

savageautomate, 2017-08-07:
I made a quick fix to what looks like the problem ... or at least part of the problem.  
https://github.com/Pi4J/pi4j/commit/7aad423a0c995d9fd57d6ab02667a92ecdd30be8

I have not tested this fix but it looked pretty obvious.  Can you please try the latest 1.2-SNAPSHOT build and test to see if this corrects the problem?  

A new snapshot build will be available in about 15 minutes.

Thanks, Robert

ribasco, 2017-08-07:
I can confirm this is fixed on the latest snapshot build. Thanks!

commit 7aad423a (2017-08-07): Fixed #353; GPIO provider was shutdown before all pins were set to their shutdown states.  In the case of an I2C based GPIO provider, the I2C bus is closed prematurely.